betaflight/betaflight #12706: No Connection after applying custom presets for IfIFLIGHT_F722_TWING

LoopLouie, 2023-04-20:
### Describe the bug

Hi, i was flashing the current Version (4.4.1) to my Iflight Succex F722 Twin G . Everthink worked fine until i applied the custom presets for the FC. After applying ic could no longer connect to the FC. It looked like i disabled USB VCP or somehting. 

I thin something in the preset for this target in not right.

When flashed Firmware 4.3.2 with the custom presets evrything works fine.

### To Reproduce

Load Firmware 4.4.1
flash to FC
Connect to FC
Popup opens and asked if custom presets want to be installed
apply custom prests

No it is not possible to connect any more.

### Expected behavior

I would have excpected that connecting to the FC after applying the custom presets is possible.

### Support ID

```txt
d5839334-ac94-40b6-8285-7ffe1f00437d
```

### Flight controller

IFlight Succex -D F7 TwingG

### Other components

CRSF nano
DJI Air unit

haslinghuis, 2023-04-20:
Please test:

[IFLIGHT_F722_TWING.zip](https://github.com/betaflight/betaflight/files/11289599/IFLIGHT_F722_TWING.zip)

haslinghuis, 2023-04-20:
Duplicate of #12635

closed 2023-04-20 by haslinghuis

LoopLouie, 2023-04-20:
I tried it but it does not solve thr problem vor me. The failure changed, but its doesn´t work either.

2023-04-20 @21:25:02 -- Chrome API Error: Failed to connect to the port..

2023-04-20 @21:25:02 -- Failed to open serial port

haslinghuis, 2023-04-20:
Restart configurator, let the FC cool off, only power from USB and retry

Also check 4.5

LoopLouie, 2023-04-20:
Okay you are right. rebootet everything and now it works like a charm. Thank you!