betaflight/betaflight #12710: Skystars H7 HD Barometer, Power, and Frsly RX not working

Some-Won-1, 2023-04-21:
### Describe the bug

1.) The baro is not recognized by the Configurator. Perhaps the template is in error for SkystarsH7HD.

2.) Also, USB does not power up GPS and receiver (s.port on RX7) until a battery is connected to the stack. Verified with 2 new stack units and checked with VOM.  I'm not sure if this is a software or hardware issue, but it makes bench configuration difficult. My other FC units don't behave this way.

3.) The Frsky ARCHER RS receiver is not able to send a bound receiver's data to the FC. Configruing with Smart Port or Frsky resets all port settings.

### To Reproduce

Plug in USB to FC and these periferals have no power and baro is inactive.

### Expected behavior

Basic USB configuration w/o battery and a barometer to use.

### Support ID

```txt
board name: SKYSTARSH7HD
manufacture id: SKST
```

### Flight controller

SkystarsH7HD stack; Racedayquads

### Other components

Frsky ARCHER RS, mode Fport
Flywoo V3 GPS-Compass

### How are the different components wired up (including port information)

GPS Mag standard default wiring.
Archer standard default wiring to RX7 (bound and set to Frsky).
Using Frsky X18S FW 1.4.8.

### Add any other context about the problem that you think might be relevant here

All new equipment and boards.

haslinghuis, 2023-04-21:
Flash core only build and provide the support ID (you can find the button in CLI tab)

Some-Won-1, 2023-04-22:
I CAN NOT get the basic core to work without a hang (red light steady). If
I do the default flash with all the suggested components, then It can boot
with flashing lights on the mobo and connect. ...But, the barometer can NOT
be detected, and the receiver port (on RX7) can NOT connect using Fport or
SmartPort bound and set with TX.

haslinghuis, 2023-04-22:
Please try 4.5.0 core build

Some-Won-1, 2023-04-23:
SkystarsH7HD 4.5.0 (zulu) now is able to configure the barometer and
magnetometer. One down and one to go. I still am unable to configure the
Frsky Archer RS to use f.port or s.port. It rejects any mode for RX in the
configurator. I read somewhere that s.port is needed for f.port telemetry,
I'm not sure if it should be set in the "port" or "receiver" section.
...And what mode is needed for the transmitter.

Some-Won-1, 2023-05-13:
Still no stable firmware. Fport not working. I need to verify the fport receiver did not get blown. Receiver has the green light though and links with TX. If you need a tester with hardware let me know. I am novas flyer.

closed 2023-06-21 by github-actions[bot]