betaflight/betaflight #12711: GEPRCF411 no Blackbox BF4.4.1  (worked on 4.3.1 build)

AndKe, 2023-04-22:
### Describe the bug

No option for blackbox on 4.4.1

*** Support data submitted *** 
Id: "1de49d3b-d1b2-43a4-b9ad-85c46f1fc8ea"

### To Reproduce

--

### Expected behavior

--

### Support ID

```txt
*** Support data submitted *** 
Id: "1de49d3b-d1b2-43a4-b9ad-85c46f1fc8ea"

FC information: https://geprc.com/product/gep-12a-f4-flight-controller-12a-esc/
```

### Flight controller

GEPRCF411

### Other components

--

### How are the different components wired up (including port information)

--

### Add any other context about the problem that you think might be relevant here

--

haslinghuis, 2023-04-22:
Did you apply custom defaults after flashing as the defines are already present: [unified target](https://github.com/betaflight/unified-targets/blob/master/configs/default/GEPR-
GEPRCF411.config)

Please try 4.5 or 4.4.0

AndKe, 2023-04-22:
no, I believed that custom defaults were NOT to be loaded when restoring a backup.

haslinghuis, 2023-04-22:
After flashing you need to apply custom defaults after first connection.
Firmware 4.5 does no longer use unified targets but a local configuration file.

closed 2023-05-26 by haslinghuis