betaflight/betaflight #12716: BF 4.4.1 Blackbox Problems

raso12, 2023-04-23:
### Describe the bug

BF 4.4.1
with the HGLRC F4V2 (OmnibusF4) I cannot select or activate the internal blackbox memory.

Thank you 
Greets Raimund

### To Reproduce

*

### Expected behavior

*

### Support ID

```txt
*
```

### Flight controller

HGLRCF4V2 (OmnibusF4)

### Other components

*

### Add any other context about the problem that you think might be relevant here

*

raso12, 2023-04-24:
ID: "f755a01b-a645-4ba8-89fa-a9b5880f10cf"

I have sent the code of the default settings (OmnibusF4) to support
The black box command and some other commands will be disabled by BF 4.4.1

raso12, 2023-04-24:
with the configurator 10.7.0 the obnibusF4 is flashed correctly and the blackbox works.

please fix the error in FlashTab Cofigurator 10.9.0

Thank you.

Greets Raimund

haslinghuis, 2023-05-25:
Fixed in: 

https://github.com/betaflight/unified-targets/commit/e8712ea2c2357f29901c7057c55121da186133da
https://github.com/betaflight/unified-targets/commit/7f1abbf7844d3299ecf2ed4cca4680dc95a941b5

closed 2023-05-25 by haslinghuis

raso12, 2023-05-25:
Thank you for fixed

Raimund