betaflight/betaflight #12723: Sending data to unpowered HDZero VTX causes FC to crash to DFU mode.

aviphysics, 2023-04-24:
### Describe the bug

Symptom:
When MSP+OSD is enabled on a uart connected to an HDZero VTX and the VTX is unpowered (such as when configuring with only USB power), the FC will be prone to crash DFU mode when switching pages in Configurator. 

This is easiest to reproduce on H7 FC, though F7 FC are also susceptible. F4 FC do not seem to fail as easily.

Cause:
HDZero VTX Rx pad impedance drops when the VTX is unpowered, which allows it to draw 22mA from the FC Tx pad when the signal is high (I have measured this via the voltage drop across a 1ohm resistor on the line from the FC Tx pad to the FC Rx pad.)

This current draw is out of spec on all STM F4, F7, and H7 MCU.

Solutions:
* User can add a 200 ohm resistor on the data line. This is small enough to not interfere with normal communication, but big enough to bring the current in spec when the VTX is unpowered. This still creates confusion for novice users, so is not a good solution for everyone 
* FC can halt sending data to VTX when VTX is not sending data to FC. When powered, the VTX polls the FC at 8Hz, which can be used to detect if VTX is powered. This is how iNav handles the situation
* Divimath could add a 200ohm resistor to their io lines, but this would not address the problem for existing hardware.

I really hate solving a hardware issue with software, but halting communications seems like the optimal solution for most users, and does not seem to terribly difficult to implement.

### To Reproduce

The easiest way to reproduce is

1) connect HDZero Rx and Gnd to respective Tx and Gnd on a FC.
2) configure FC for MSP+VTX on that uart
3) power FC with USB
4) observe that red LED on HDZero VTX dimly glows, indicating that it is drawing power from the FC Tx pin
5) If using an H7 or F7 MCU on FC, connect with configurator and cycle through configuration pages.
6) Observe that FC will crash to DFU mode (seems most likely to happen on H7 MCU.)

Place 200ohm resistor between FC Tx pad and VTX Rx pad and repeat. Observe that FC does not crash.

You can also repeat without the resistor and with uart disabled and FC will not crash.

### Expected behavior

FC does not crash

### Support ID

```txt
I don't have this on me. Please see description.
```

### Flight controller

Any with an H7 MCU is easiest to test on. F7 is also affected.

### Other components

Any HDZero VTX

### How are the different components wired up (including port information)

HDZero VTX connected to uart configured for MSP+OSD

### Add any other context about the problem that you think might be relevant here

See description

haslinghuis, 2023-04-24:
@SteveCEvans any time to spin up:

> FC can halt sending data to VTX when VTX is not sending data to FC. When powered, the VTX polls the FC at 8Hz, which can be used to detect if VTX is powered. This is how iNav handles the situation

haslinghuis, 2023-04-24:
@SteveCEvans any time to spin up a LBT solution:

> FC can halt sending data to VTX when VTX is not sending data to FC. When powered, the VTX polls the FC at 8Hz, which can be used to detect if VTX is powered. This is how iNav handles the situation

SteveCEvans, 2023-04-25:
Can you please check if this works OK if connected to USARTs other than 1 or 3.

It’s the RX line that’s monitored by the FC to potentially enter DFU mode, not TX. What you’ve done is effectively add a pull-up on the RX line. You should do that from the 3V3 line, not the TX line.

aviphysics, 2023-04-25:
> Can you please check if this works OK if connected to USARTs other than 1 or 3.
> 
> It’s the RX line that’s monitored by the FC to potentially enter DFU mode, not TX. What you’ve done is effectively add a pull-up on the RX line. You should do that from the 3V3 line, not the TX line.

I don't understand what you are asking.

The Tx pad communicates to the VTX by alternating the FC Tx pad between Gnd and 3.3v. The problem is that the impedance from the VTX Rx pad to Gnd drops when the VTX is unpowered. This creates a load of ~22mA on the FC Tx pad, which is out of spec for the MCU and makes the MCU unstable.

On iNav, we found that halting sending data to the VTX when the VTX is unpowered fixed the problem. We can know when the VTX is unpowered because the VTX sends data to the FC Rx pad at 8hz when it is powered.

aviphysics, 2023-04-25:
When I say, add resistor on the data line, this is what I mean.
[image]

SteveCEvans, 2023-04-25:
Could you please try removing the resource for the TX pin and instead setting it to be PINIO configured to output a low. That will prevent current being drawn on the TX line.

aviphysics, 2023-04-25:
> Could you please try removing the resource for the TX pin and instead setting it to be PINIO configured to output a low. That will prevent current being drawn on the TX line.

Hi Steven, thank your for engaging with the problem.

Personally, I spent a solid week of my life diagnosing and testing this problem to develop the fix for iNav. I have written what I already know. Someone else needs to do the testing for BF.

howels, 2023-04-26:
> Hi Steven, thank your for engaging with the problem.
>
> Personally, I spent a solid week of my life diagnosing and testing this problem to develop the fix for iNav. I have written what I already know. [...]

Please link to the iNav issue for more data

SteveCEvans, 2023-04-26:
I've tried reproducing this using both a ZEEZF7V3 and a SPEEDYBEEF7V2 connected to an unpowered WS VTX and don't see this issue. Is this only seen with the HD Zero VTX. Note that both of these FCs use USART2 for the VTX. I suspect that only USART1 and USART3 may be affected.

aviphysics, 2023-04-26:
> I've tried reproducing this using both a ZEEZF7V3 and a SPEEDYBEEF7V2 connected to an unpowered WS VTX and don't see this issue. Is this only seen with the HD Zero VTX. Note that both of these FCs use USART2 for the VTX. I suspect that only USART1 and USART3 may be affected.

You would need to test with an HDZero VTX.

The problem I believe is due to the impedance to ground on the Divimath chip io pins dropping when the chip is not powered.

aviphysics, 2023-04-26:
> Please link to the iNav issue for more data

I had queried Geoff for a link to the pull request that contains the fix, but he is busy getting married.

I had discovered the issue before the code was released and worked with Geoff to test his fix, so there was never a bug report in the iNav code.

howels, 2023-04-26:
It could be in here but I can't see an obvious commit
https://github.com/iNavFlight/inav/pulls?q=is%3Apr+author%3Ageoffsim+

Sounds like an HDzero hardware issue, and I've personally run HDzero whoop VTX and not encountered the error.
We should be talking to HDzero about this, get it fixed in hardware as it sounds like a hardware bug.

SteveCEvans, 2023-04-26:
Please confirm the F7 USART which has been demonstrated to show this issue.

aviphysics, 2023-04-27:
> It could be in here but I can't see an obvious commit https://github.com/iNavFlight/inav/pulls?q=is%3Apr+author%3Ageoffsim+
>
> Sounds like an HDzero hardware issue, and I've personally run HDzero [...]

I agree that it would be best to have it fixed in hardware, but there is no way Divimath is going to replace every VTX that is already out there. Even if they wanted to, the expense would kill the company.

I think there is also a reasonable argument that FC shouldn't be released with unprotected uarts, and it could be called a fault with FC design.

Ultimately, the solution to just not send data when the VTX is unpowered is a relatively easy solution.

aviphysics, 2023-04-27:
> Please confirm the F7 USART which has been demonstrated to show this issue.

On the KakuteF7mini v3 I was able to reproduce on every uart when I did the testing with iNav.

H7 MCU seem even more prone to crashing from this, so I would recommend using an H7 for testing.

It is also possible that your FC was built with resistors on the io lines, to prevent this type of failure.

aviphysics, 2023-04-27:
> It could be in here but I can't see an obvious commit https://github.com/iNavFlight/inav/pulls?q=is%3Apr+author%3Ageoffsim+
>
> Sounds like an HDzero hardware issue, and I've personally run HDzero [...]

I was going to bug Geoff in a couple weeks, after which maybe his Honeymoon will be over and he will have time to help out.

aviphysics, 2023-04-27:
The code for iNav that suspends sending data to the VTX is somewhere in here.
https://github.com/iNavFlight/inav/pull/7668

haslinghuis, 2023-04-27:
```
/*
 * Intercept MSP processor.
 * VTX sends an MSP command every 125ms or so.
 * VTX will have be marked as not ready if no commands received within VTX_TIMEOUT.
 */
static mspResult_e hdZeroProcessMspCommand(mspPacket_t *cmd, mspPacket_t *reply, mspPostProcessFnPtr *mspPostProcessFn)
{
    if (vtxSeen && !vtxActive) {
        vtxReset = true;
    }

    vtxSeen = vtxActive = true;
    vtxHeartbeat = millis();

    // Process MSP command
    return mspProcessCommand(cmd, reply, mspPostProcessFn);
}
```

ledvinap, 2023-04-28:
It may be better to simply check VTX RX pin. It  is normally high (unless data bit is sent) and will be LOW when VTX is powered off.
I can think of some possible tests:
 - detecting BREAK condition in serial driver 
 - testing RX level + checking if receive is in progress
 - waiting for VTX RX high level before setting vtxActive (and clearing it after few low samples)
 - sending VTX command only when VTX RX is high (it will also partially help with half duplex if RX and TX are connected), with retry after small delay

SteveCEvans, 2023-04-28:
I have a plan for this, but please give me time to get to it as I’m working on another issue,

In short my plan is to initialise USART outputs by default as GPIO inputs with an internal pull up. Then, whenever an attempt is made to transmit data if no data is currently being transmitted I’ll confirm that the line isn’t being pulled low, and then if not, enable the TX output and send the data. Then, once the last byte is transmitted, I’ll switch it back to an input. In this way we’ll also detect when a device is turned off. This will directly address the assertion that too much current is being drawn from the TX line.

Further, to help prevent inadvertent DFU issues I may also need to do similar on the RX line pin configuration based on the TX config.

ledvinap, 2023-04-28:
@SteveCEvans : There is no strict requirement on TX pin loading now. Everything works even when TX pin is loaded to GND (so that pin pullup won't get it over high threshold). 
Testing pin with pullup may break some configurations ..

aviphysics, 2023-04-28:
> It may be better to simply check VTX RX pin. It is normally high (unless data bit is sent) and will be LOW when VTX is powered off. I can think of some possible tests:
>
> * detecting BREAK condition [...]

It is interesting. I suspect the DC impedance is higher than the impedance when data is being sent. I had noticed that it was normal high.

The impedance of the VTX is definitely not low enough to drop the voltage to ground at the FC Tx pin. I don't think that would work.

aviphysics, 2023-04-28:
> I have a plan for this, but please give me time to get to it as I’m working on another issue,
>
> In short my plan is to initialise USART outputs by default as GPIO inputs with an internal pull up. [...]

The DC impedance of the MCU isn't low enough to drag the FC Tx pin to ground.

In all my testing, the Rx pin on the FC was never a problem. The FC Rx pin isn't trying to drive voltage, so it doesn't have the problem the FC Tx pin does.

I appreciate the creativity, but I think you are overthinking the solution.

The iNav solution really does work.

aviphysics, 2023-05-01:
Today I visited a local with the crash to DFU problem on an H7 FC. Adding a 200 ohm resistor in series with the FC Tx pad and VTX Rx pad fixed the crash to DFU problem.

The resistor is hidden under the FC, but it is wired like this.

The point of the resistor is that the VTX Rx pad impedance to ground is a little low when the VTX is unpowered and this makes the FC unstable when it is sending data to the unpowered VTX.

The resistor adds enough impedance to keep the current draw from the FC within spec.

The H7 flight controllers seem the most sensitive to the current draw, though it seems like some F7 and F4 FC are also affected.

[image]

aviphysics, 2023-05-02:
I was also able replicate the DFU crash to DFU with the FC on USB power, FC configured for MSP+OSD, and HDZero VTX attached.

This time with a whoop VTX and Holybro Kakute H7 Mini V1.3.

The resistor fix also worked in this case.

I did this test under BF4.4.1 and BF4.4. Though BF4.4 also has the problem, for some reason BF4.4.1 is even more prone to crashing to DFU.

[image]

SteveCEvans, 2023-05-04:
Please test https://github.com/betaflight/betaflight/pull/12760/files which prevents a powered down device from drawing excessive current from a TX line.

Please test, checking the sensors tab (and without any debug mode selected) and you should see if the device connected to a port is unpowered and dragging down the TX line.

For example, in the below an ELRS RX is connected to UART 1, and a WS VTX to UART 4 being turned on and then off again.

[image]

SteveCEvans, 2023-05-05:
Debug now removed. I'd appreciate somebody testing on an HD Zero VTX. I've been able to demonstrate the DFU issue using WS and that the PR fixes this issue.

SteveCEvans, 2023-05-09:
@aviphysics  could you please test the PR.

closed 2023-05-10 by sugaarK

aviphysics, 2023-05-10:
> @aviphysics could you please test the PR.

Would you mind sending me a version compiled for a H745 target? (Specifically Holybro Kakute H7)

aviphysics, 2023-05-11:
> @aviphysics for testing: in configurator choose show release candidates and enable expert mode. Select development releases and auto-detect your board.

Cool... I should have time in the next couple days.

aviphysics, 2023-05-11:
Under the released 4.4.1, I can't get the original error to reproduce on the H743 board I currently have available for testing.

The board I original reproduced the error on is currently being used for some Ardupilot development, and I would have to tear down that quad to remove the resistor mod too.

I'm not sure when I will have time to really dig in with an oscilloscope and verify that the fix is doing what we think it ought to be doing.

haslinghuis, 2023-05-11:
This PR is only in 4.5.0 right now and will be available in 4.4.2

aviphysics, 2023-05-12:
> This PR is only in 4.5.0 right now and will be available in 4.4.2

I understand it isn't in 4.4.1, but I was trying to make sure I could reproduce the original problem, so I could tell that this fixed it.

Mostly-Harmless-420, 2024-08-10:
Hopefully this isn't too long winded, but I'm experiencing something that appears to be related to this issue. FWIW I don't have an oscilloscope, I'm not an electrical engineer- just thought i'd share my findings if they're relevant/helpful. 

Today i set up a new quad: HappyModel Super X HD FC + HDZero Whoop Lite VTX

Initially i was powering the VTX with vbat - i discovered quickly that, while connected to BetaFlight via USB, attaching a battery (to test/configure motors) Betaflight would abruptly go into DFU mode after a few seconds.

The board was originally on 4.4.2 - i updated to 4.5.1 ... The same issue persisted.

The quad flies fine with the HappyModel Mobula 6 Eco presets - everything seems to work properly _excluding battery power while USB is connected._

I was suspicious about the VTX because i hadn't had problems when originally testing the naked FC - so i disconnected it from vbat/ground ... Connected to Betaflight and plugged in a battery ... No issues (no sudden DFU mode) ... Basically confirming my VTX is triggering DFU. (So that's how i ended up here via Google "VTX triggers DFU mode betaflight" lol)

Now, just to see, i wired it up to the 5v pad near my rx/tx pads ... Now my problem is half resolved and kinda inverted:

When i plug in USB to the unpowered quad BF instantly goes into DFU mode. However if i plug in a battery first and then USB, everything seems to work normally (like all my other quads), I've not observed it flipping to DFU in this use case.

Is it safe to say a 200ohm resistor on the TX would resolve this? Or maybe just ignore it and remember to plug a battery in before using BF? 🙃

Or can/will this be resolved in future BF version?

SteveCEvans, 2024-08-10:
This should be fixed by https://github.com/betaflight/betaflight/pull/12760 and a number of related PRs which followed. I'll check this out.