betaflight/betaflight #12726: JHEMCU GHF411AIO-ICM 40A F411 ICM42688P Flight Controller: No accelerometer & gyro after upgraded to 4.4.1

Canaill51, 2023-04-25:
### Describe the bug

Updated from 4.3.1 to 4.4 and no accelerometer was detected.

board: manufacturer_id: JHEF, board_name: JHEF411

### To Reproduce

Flash from 4.3.1 to 4.4.1

FYI,
I tried to flash it on 4.5-zulu prerelease : Accelerometer and gyro were detected.

### Expected behavior

Accelerometer and gyro should be detected in 4.4.1

### Support ID

```txt
d1c9cdee-c938-44e3-9f3b-d205e589e472
```

### Flight controller

JHEMCU GHF411AIO-ICM 40A F411 ICM42688P

haslinghuis, 2023-04-25:
For 4.4.1 you can add `GYRO_SPI_ICM42688P ACC_SPI_ICM42688P` in custom defines before flashing again?

Canaill51, 2023-04-25:
> For 4.4.1 you can add `GYRO_SPI_ICM42688P ACC_SPI_ICM42688P` in custom defines before flashing again?

@haslinghuis , this time, it has fixed the issue. I previously tried also to add these in Custom Defines, but did not work.

Did you make any changes since my incident?

Anyway, thanks :)

haslinghuis, 2023-04-25:
To use custom defines you need either nightly build of configurator or enable show release candidates and show expert mode. Without this field is ignored in the build.

Canaill51, 2023-04-25:
> To use custom defines you need either nightly build of configurator or enable show release candidates and show expert mode. Without this field is ignored in the build.

Got it! Thanks, I was not sure about this.

haslinghuis, 2023-04-25:
4.5 is pretty stable at the moment using nightly configurator. As we are finalizing migration from unified targets to local configuration - it would be possible but requires more backports from 4.5 into 4.4.2.

exviia, 2023-05-11:
Hello.. i flash from 4.3.1 to 4.4.1.. and from the second-third time  gyroscope and accelerometer appear.. 

what I can't get to work is softserial, I put smartaudio on led strip in 4.3.1. ..and i use uart1 for gps but when i go to 4.4. - there is no led_strip in resource.. and gps doesn't work,,

Canaill51, 2023-06-01:
@haslinghuis ,
FYI, 4.4.2 has fixed this issue :) Thanks for your support, as always.

closed 2023-06-01 by Canaill51