betaflight/betaflight #12737: Faulty GPS rescue on modern ublox M10 GPS

shangooz, 2023-04-28:
### Describe the bug

Modern ublox M10 GPS can work and sometimes default to higher GPS update rates such as 15 Hz or 16.6667 Hz (60ms).

[image]

Betaflight currently supports a maximum update rate of 10 Hz, so using 15 Hz or 16.667 Hz GPS causes wild twitching on the pitch axis.

[image]

Video of the behaviour: https://user-images.githubusercontent.com/78979329/235086448-2efdd98b-b20d-4a5e-8764-f059ba830b1c.mp4

Autoconfig and autobaud currently do not configure M10 GPS properly.

Logfiles with autoconfig on and autoconfig + autobaud on.
[BBL autoconfig.zip](https://github.com/betaflight/betaflight/files/11351156/BBL.autoconfig.zip)

Currently the only solution is to disable autoconfig and manually set the M10 GPS to 10 Hz using U-Center. Without autoconfig, dynamic models will not be set, HDOP display and other things will not work. GPS Rescue itself will work though.

### To Reproduce

Install M10 GPS with high update rate to a BF 4.4.1 quad and autoconfig GPS and GPS Rescue.

### Expected behavior

Normal function of GPS Rescue as with other M8 GPS.

### Support ID

```txt
59de8267-3ecc-4da0-804b-8677a0d09700
```

### Flight controller

iFlight Blitz F7 mini

ZzyzxTek, 2023-04-29:
Betaflight uses the UBX-CFG-RATE message to set the measurement rate to 100ms (10Hz) and the solution ratio to 1:1.

UBX-CFG-RATE is supported on the protocols for the M8 and below (protocol versions thru 23.01).

The M9 spec seems to still support UBX-CFG-RATE, but says "This message is deprecated in protocol versions greater than 23.01. Use UBX-CFG-VALSET, UBX-CFGVALGET, UBX-CFG-VALDEL instead."

Taking an initial look at the M10 protocol spec, they appear to have changed to using a generic UBX-CFG-VALSET message that takes config key/value sets (e.g., CFG-RATE-MEAS = 100 and CFG-RATE-NAV = 1).

If that's the case, and the M10 no longer supports UBX-CFG-RATE, BF would need to implement M10 specific code to support the UBX-CFG-VALSET message.

I would think on an M10, sending a UBX-CFG-RATE message should get a NAK reply, but the code seems to have very limited handling of rejected config settings.

@shangooz Since you already are connecting via u-center, would you please post the output of the UBX-MON-VER message? I'd like to see the protocol version it reports. Thanks!

McGiverGim, 2023-04-29:
@ASDosjani you have fight against the GPS code recently. Do you think will be difficult to use different codes depending on the protocol version detected?

ASDosjani, 2023-04-29:
I think it could be implemented. Unfortunately l'm fighting against the time recently but I write it up to my list.
First of all, as @ZzyzxTek said, I'd like to see the protocol version.

McGiverGim, 2023-04-29:
Of course, don't worry.
I link here two iNav PRs that can help, for the future:
https://github.com/iNavFlight/inav/pull/8058
https://github.com/iNavFlight/inav/pull/6269

ZzyzxTek, 2023-04-29:
I've been looking at INAV, Ardupilot, PX4 to see how they handle it.

There isn't a whole lot of variance between M8, M9, M10 protocol versions, especially considering the limited subset BF uses right now. Mostly in the config, also a few commands like NAV-PVT instead of NAV-SOL.
INAV, for example, bases choosing which protocol on the hardware version (M8, M9, M10). The protocol variations within a hardware version don't deprecate anything. It might be more robust to check protocol version, but it's a bit more work to parse that out of the UBX-MON-VER payload.
If BF wants to limit backward compatibility, for example, nothing before M8 supported, that would help.

I think it could be initially added into the current gps.c code with some conditional logic, at first just targeting the minimal things (like nav rate) that we want to fox first. Might get a little messy.
imho, to do this "right" we'd want to refactor the state machine a bit, and make the handling of ACK/NAK a bit more robust. Not crazy hard, but certainly not a quick effort either, especially considering testing.

closed 2023-08-12 by blckmn