betaflight/betaflight #12739: GPS Rescue 3D Mode

KenImhof, 2023-04-28:
### Is your feature request related to a problem? Please describe

GPS Rescue is grayed-out and not currently available in version version 4.4 when 3D mode is enabled.

### Describe the solution you'd like

Add support for GPS rescue in 3D mode.

### Describe alternatives you've considered

Allow GPS rescue when 3d disable / switch is enabled.

haslinghuis, 2023-04-28:
Ref: https://github.com/betaflight/betaflight/issues/9233

ctzsnooze, 2023-08-14:
To fix this would require code that temporarily disabled 3D mode when GPS Rescue is enabled, instead of the current behaviour, which blocks GPS Rescue when 3D mode is enabled.  

So far no-one has made a PR to do this; the code is extremely complex.

chimera15, 2023-10-25:
This should be a simple fix to prevent rth in when 3d is enabled  in flight, not in the configurator.  This is wrong, you're preventing me from safely flying up the side of a mountain away from me using gps, then flying back upside down if I want because I know I have range to do so.

closed 2026-09-26 by blckmn