betaflight/betaflight #12742: HobbywingF7 Convertible different Gyro

KCfpv, 2023-04-29:
### Describe the bug

The hobbywing f7 convertible fc comes with an mpu6500 gyro and is no longer being sold with the icm gyro. I suggest possibly creating two different cloud builds with different gyros, as I can not access kaack mod since I need to flash core only.

### To Reproduce

Flash an updated version of the hobbywing f7 convertible fc

### Expected behavior

It will not detect a gyro.

### Support ID

```txt
cb7550b3-5905-4920-bb62-9d75db5097a5
```

### Flight controller

Hobbywing f7 convertible bought from hobbywing direct

### Other components

N/A

Gozzz, 2023-05-06:
Any fix for this on 4.4? 4.5 seems to work but it is flying quite twitchy. I'm running elrs 250hz and I get a warning when applying the preset in 4.5, not sure if that is the cause of the twitch or not?

KCfpv, 2023-05-06:
> Any fix for this on 4.4? 4.5 seems to work but it is flying quite twitchy. I'm running elrs 250hz and I get a warning when applying the preset in 4.5, not sure if that is the cause of the twitch or not?

When flashing follow the highlighted steps

[image]

closed 2023-06-01 by haslinghuis