betaflight/betaflight #12744: GEPRC_F722_AIO: External barometer GEP-M8Q not working after update to BF 4.4.1

mrtxilya, 2023-04-29:
### Describe the bug

I have an external module GEP-M8Q (GPS + MAGNETOMETER + BARO) [PDF Link](https://geprc.com/wp-content/uploads/2023/04/GEP-M8Q-GPS%E8%AF%B4%E6%98%8E%E4%B9%A6.pdf)
GPS connected to UART5; SDA and SCL connected to R3, T3.
On version 4.3.2 everything works well.
[image]

After updating to 4.4.1, I could not turn on the barometer option, after clicking on the "Save and Reboot" the option was reset to off.
I flashed the controller again with the `BARO BARO_BMP280` Custom Defines, after that I was able to enable the barometer option, but it does not work.
But at the same time, the magnetometer works well.
[image]

### To Reproduce

Update GEPRC_F722_AIO with external module GEP-M8Q to BF 4.4.1 with the `BARO BARO_BMP280` Custom Defines.

### Expected behavior

The barometer works just as well as the magnetometer.

### Support ID

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr 28 2023 / 22:02:59 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: GEPR, board_name: GEPRC_F722_AIO

# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=55degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3043, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P, MAG=QMC5883
OSD: MSP (53 x 20)
BUILD KEY: 34865d749f2a895668eed4369c25e21e (4.4.1)
System Uptime: 14 seconds, Current Time: 2023-04-29T12:44:39.115+00:00
CPU:59%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 2203 * 0.01V (6S battery - OK)
I2C Errors: 6
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP GPS

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr 28 2023 / 22:02:59 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRC_F722_AIO
manufacturer_id GEPR

# resources
resource BEEPER 1 C15
resource MOTOR 1 C07
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C08
[... 325 lines cut ...]
blackbox_disable_mag = OFF
Allowed values: OFF, ON

gps_rescue_use_mag = ON
Allowed values: OFF, ON

# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 2
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388, DPS310, 2SMPB_02B

altitude_prefer_baro = 100
Allowed range: 0 - 100
```

haslinghuis, 2023-04-29:
> SDA and SCL from MAG/BARO connected to R3, T3.

[image]

D / C should be connected with SDA /SCL respectively but I can't find SDA / SCL on the FC diagram:
https://geprc.com/wp-content/uploads/2022/05/F722-35A-AIO-USER-MANUAL2.pdf
https://geprc.com/wp-content/uploads/2022/10/F722-45A-AIO-USER-MANUAL.pdf

GEP M8Q: Communication Protocol: UBLOX(GPS)、I2C(Mag&Baro)

mrtxilya, 2023-04-29:
On version 4.5 I can't activate the barometer. After saving and reloading the option is disabled.
I can't build firmware with custom defines `BARO BARO_BMP280`. Cloud build failed.
https://build.betaflight.com/api/builds/d836918fcedf8a0cbace543459add3d5/log

haslinghuis, 2023-04-29:
Try using BARO_BMP280 in the define (without BARO)

mrtxilya, 2023-04-29:
Cloud build with the BARO_BMP280 define was successfull. After flashing I still can not activate the barometer option. After saving and reloading the option is disabled.

haslinghuis, 2023-04-29:
Please check if resources are preserved
Output of `resource show all` and `get baro`

haslinghuis, 2023-04-29:
:boom: try BARO_MS5611

mrtxilya, 2023-04-29:
With `BARO_MS5611` barometer is active. But it shows 0 altitude all the time on BF 4.5 and on BF 4.4.1.
Tried to `set baro_hardware = MS5611` but it didn't help.
[image]

```
# get baro
baro_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

baro_spi_device = 0
Allowed range: 0 - 5

baro_i2c_device = 2
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

baro_hardware = MS5611
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280, LPS, QMP6988, BMP388, DPS310, 2SMPB_02B
Default value: AUTO

altitude_prefer_baro = 100
Allowed range: 0 - 100
```

haslinghuis, 2023-04-29:
Unrelated but instead of the option Magnetometers you also could add custom define: MAG_QMC5883
Baffled why BARO is not updating (onboard does).

mrtxilya, 2023-04-30:
I rechecked baro on BF 4.3.2 and it works, so there are no hardware issues with baro.
[image]

Baro stops working from version 4.4.0.

Also I tried only `MAG_QMC5883` custom define and with it I couldn't activate magnetometer.
Magnenometer works with `MAG MAG_QMC5883` custom defines.

My current build 4.4.1 with `MAG MAG_QMC5883 BARO BARO_MS5611` custom defines and mag, baro are active but still 0 altitude all the time.

Neolight79, 2023-05-11:
I faced the same issue. GEP-M8Q with MS5611 baro onboard does not produce any data despite MS5611 I2C device is available and works well with 4.3.2. Tried 4.5.0 - no luck. Magnetometer QMC5883 works well on the same I2C line. Obviousely, problem is in the MS5611 driver.

haslinghuis, 2023-05-11:
Did you try a core only build (for testing)?

Neolight79, 2023-05-11:
> Did you try a core only build (for testing)?

Yes, the result is the same - barometer data is always 0.

Neolight79, 2023-05-11:
Tested all the FW from 4.4.0 (4.4.0, 4.4.1, 4.4.2, 4.5.0, with CORE ONLY and with custom defines) - all have the same problem, barometer data is not changing. With 4.3.2 everything is OK.

haslinghuis, 2023-05-11:
To narrow it done further you can flash these:

https://github.com/betaflight/betaflight/releases/tag/4.4.0-RC3
https://github.com/betaflight/betaflight/releases/tag/4.4.0-RC2
https://github.com/betaflight/betaflight/releases/tag/4.4.0-RC1

Neolight79, 2023-05-12:
Well, installed 4.3.2 again - baro works well, the next release 4.4.0 RC1:
Betaflight / STM32F7X2 (S7X2) 4.4.0 Dec 14 2022 / 03:39:14 (103f238) MSP API: 1.45 (installed it from HEX because it is not available in the list)
Baro stops working. The problem arose between 4.3.2 and 4.4.0 RC1.

haslinghuis, 2023-05-12:
@KarateBrot could the (higher) task rate be the issue here?

Neolight79, 2023-05-12:
I'm available for gatherning any source information, conduct the tests and experimants. I already created very nice holder for this module and using it without baro is very undesirable.
[image]

KarateBrot, 2023-05-12:
> @KarateBrot could the (higher) task rate be the issue here?

No, that should not be a problem at all. You would at least see a lot of late BARO task runs when typing `tasks`.

Neolight79, 2023-05-12:
> > @KarateBrot could the (higher) task rate be the issue here?
> 
> No, that should not be a problem at all. If the task had problems running, you would at least see a lot of late BARO task runs when typing `tasks`.
> 
> Do we know if all baros run over I2C without problems? Unfortunately I don't have a GPS module with integrated baro to test anything here. (Need to double check when I get home on Sunday)

Let me know, please, if I can conduct any tests to provide you with additional info.

haslinghuis, 2023-05-12:
@Neolight79 try:  [betaflight_4.5.0_STM32F7X2_GEPRC_F722_AIO.zip](https://github.com/betaflight/betaflight/files/11463035/betaflight_4.5.0_STM32F7X2_GEPRC_F722_AIO.zip)

KarateBrot, 2023-05-12:
Altitude shows 0 unless you are armed and flying

Neolight79, 2023-05-12:
Concerning 0 unless arming. I thought it over before, but I have another quad with built in DPS310 on 4.4.0 - it shows altitude before drone armed, Anyway, I'll try another test with arming.
Tasks output attached.
[tasks.txt](https://github.com/betaflight/betaflight/files/11463739/tasks.txt)

Neolight79, 2023-05-12:
Is it possible al all to arm quad while it has USB connection? I'll always have MSP flag.

haslinghuis, 2023-05-12:
Yes - REMOVE PROPS - and enable motor test (at least works with configurator nightly build)

haslinghuis, 2023-05-12:
What's the value of `get altitude_prefer_baro` and `get altitude_source` ?

mrtxilya, 2023-05-12:
Hello.
In my case, when the baro is on, 0 altitude is shown all the time, even during the flight (in telemetry and OSD)
I had to turn off the baro so that the altitude is displayed in the OSD (from GPS)

```
# get altitude
telemetry_disabled_altitude = OFF
Allowed values: OFF, ON

osd_altitude_pos = 19619
Allowed range: 0 - 65535
Default value: 341

altitude_source = DEFAULT
Allowed values: DEFAULT, BARO_ONLY, GPS_ONLY

altitude_prefer_baro = 100
Allowed range: 0 - 100

altitude_lpf = 300
Allowed range: 10 - 1000

altitude_d_lpf = 100
Allowed range: 10 - 1000
```

Neolight79, 2023-05-12:
> What's the value of `get altitude_prefer_baro` and `get altitude_source` ?

get altitude_prefer_baro = 100
altitude_source = DEFAULT (tried BARO_ONLY - no result)
Also cleared all arming disable flags - no result, altitude is always 0.

Neolight79, 2023-05-12:
I made the experiment. I installed FW 4.5.0, set DEBUG for BlackBox to BARO mode and made some movements up and down of the armed drone. After that I repeated this but with 4.3.2 onboard:
[image]
I discovered that BaroState changing normally 
[BBL logs.zip](https://github.com/betaflight/betaflight/files/11466075/BBL.logs.zip)
on both versions. BaroTemperature is constant 999 and not changing on 4.5.0, BaroPressure on 4.5.0 looks like BaroTemp on 4.3.2 and it is rising slowly (only grows despite I move drone up and down), BaroSum is 0 on 4.5.0 as expected. Logs attached.

Neolight79, 2023-05-12:
Another interesting fact. I conducted the same test on working drone with DPS310 baro (no problem on 4.4.0) but I foud out that DEBUG_BARO data is similar to MS5611 on 4.5.0:
[image]
Baro state cycle works well but baro temperature is constant 995 and Baro Pressure growing thru the log (the same behavior as MS5611 in 4.5.0). But in this case Baro Pressure Sum shows real data according the drone's movings. I suppose that debug data is wrong on 4.4+ FW but for some reason counting of altitude is correct for DPS310.
Just noticed, in 4.3.2 BARO_DEBUG1=Temp and BARO_DEBUG2=Pressure, but for 4.4.0 BARO_DEBUG1=Pressure and BARO_DEBUG2=Temp. Anyway, can't think anymore today)

Neolight79, 2023-05-14:
Well, concerning DEBUG values. In 4.4.0 DEBUG values for BARO changed: Temp&Press got new place (I wrote in last post) and also values become floats not int as in 4.3.2. I've made the test: I changed the DEBUG values back (only for debug purposes) and conducted the same test as on Friday.
[image]
In fact Temperature and Pressure look the same as on 4.3.2, but resulting value is always 0.
This means that data from barometer are gathered but not calculated correctly.

Neolight79, 2023-05-14:
One more experiment:
[image]
1) We know that MS5611 provides data. Temp&Press data is the same for 4.3.2 and 4.4.1 but result is 0 in 4.4.1.
2) I made records to blackbox from MS5611 and DPS310 both running 4.3.2. Temp and Press are aproximately similar, but PressSum are absolutely different. Thus, we can see that resulted altitude counting differently on the same fw 4.3.2 but for different ICs.
[image]

haslinghuis, 2023-05-14:
[betaflight_4.5.0_STM32F7X2_GEPRC_F722_AIO.zip](https://github.com/betaflight/betaflight/files/11473153/betaflight_4.5.0_STM32F7X2_GEPRC_F722_AIO.zip)

Neolight79, 2023-05-15:
> [betaflight_4.5.0_STM32F7X2_GEPRC_F722_AIO.zip](https://github.com/betaflight/betaflight/files/11473153/betaflight_4.5.0_STM32F7X2_GEPRC_F722_AIO.zip)

Unfortunately, with this fw also does not work. After applying my config also loosing USB device (maybe because MAG feature was not included into the build) but this is not the point. BARO available but altitude is still 0.

Neolight79, 2023-05-15:
My findings for today for 4.4.1:
1) If move writing to DEBUG_BARO before baroIsCalibrated() check (\src\main\sensors\barometer.c (428)) then we can read "baro.altitude" alternating according moving up and down.
```
            // update baro data
            baro.dev.calculate(&baro.pressure, &baro.temperature);
            baro.altitude = pressureToAltitude(baro.pressure);

            // New place
            DEBUG_SET(DEBUG_BARO, 1, lrintf(baro.pressure / 100.0f));   // hPa
            DEBUG_SET(DEBUG_BARO, 2, baro.temperature);                 // c°C
            DEBUG_SET(DEBUG_BARO, 3, lrintf(baro.altitude));            // cm

            if (baroIsCalibrated()) {
                // zero baro altitude
                baro.altitude -= baroGroundAltitude;
            } else {
                // establish stable baroGroundAltitude value to zero baro altitude with
                performBaroCalibrationCycle(baro.altitude);
                baro.altitude = 0.0f;
            }

            // Old place
            //DEBUG_SET(DEBUG_BARO, 1, lrintf(baro.pressure / 100.0f));   // hPa
            //DEBUG_SET(DEBUG_BARO, 2, baro.temperature);                 // c°C
            //DEBUG_SET(DEBUG_BARO, 3, lrintf(baro.altitude));            // cm
```
2) In 4.4.1 "DEBUG_SET(DEBUG_BARO, 3, ..." is writing twice (\src\main\sensors\barometer.c (439) & \src\main\flight\position.c (212)), that is why we can't see all the values for baro.altitude during whole the period - 0 is writing in "position.c", but in "barometer.c" (according to point 1) we can write alternating altitude values.

haslinghuis, 2023-05-15:
Nice, had the same idea yesterday - so that works !

Next part to consider is baroIsCalibrated():

            if (baroIsCalibrated()) {
                // zero baro altitude
                baro.altitude -= baroGroundAltitude;
            } else {
                // establish stable baroGroundAltitude value to zero baro altitude with
                performBaroCalibrationCycle(baro.altitude);
                baro.altitude = 0.0f;
            }

Neolight79, 2023-05-16:
No success so far.
If decrement ```baro.altitude -= baroGroundAltitude;``` is commented, I can get current measured altitude in baro.altitude which is showing correctly in global altitude value and in debug value as well. But in case I'm starting any counting and assigning new value to baro.altitude - it becomes 0.

Neolight79, 2023-05-16:
For the moment I found out that ```static void performBaroCalibrationCycle(const float altitude)``` works incorrectly (for MS5611). For sure, problem is not in this procedure. I guess problem could be in the process of gathering and counting values during baro calibration.

Neolight79, 2023-05-16:
The last thing I finally found (but this not resolves the problem in general):
This works (of course baroGroundAltitude is not avareged but taken the last one, actually this is the only change for whole build):
```
static void performBaroCalibrationCycle(const float altitude)
{
    static uint16_t cycleCount = 0;

    baroGroundAltitude += altitude;
    cycleCount++;

    if (cycleCount >= calibrationCycles) {
       // baroGroundAltitude /= cycleCount;  // simple average
        baroGroundAltitude = altitude;  // only last sample
        baroCalibrated = true;
        cycleCount = 0;
    }
}
```
But this does not work (always 0):
```
static void performBaroCalibrationCycle(const float altitude)
{
    static uint16_t cycleCount = 0;

    baroGroundAltitude += altitude;
    cycleCount++;

    if (cycleCount >= calibrationCycles) {
        baroGroundAltitude /= cycleCount;  // simple average
       // baroGroundAltitude = altitude;  // only last sample
        baroCalibrated = true;
        cycleCount = 0;
    }
}
```

ledvinap, 2023-05-16:
Most probable cause of this issue is that `performBaroCalibrationCycle()` is called with NaN (or, unlikely, Inf) as parameter.

This is quite big problem - `-ffast-math` optimization implies [-ffinite-math-only](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html#index-ffinite-math-only), so a lot of bad things can happen.
https://simonbyrne.github.io/notes/fastmath/

@Neolight79:

Can you try:
```
__attribute__((optimize("-fno-finite-math-only"))) 
static void performBaroCalibrationCycle(const float altitude)
{
    static uint16_t cycleCount = 0;
    if (altitude != altitude) return;

    baroGroundAltitude += altitude;
    cycleCount++;

    if (cycleCount >= calibrationCycles) {
        baroGroundAltitude /= cycleCount;  // simple average
        baroCalibrated = true;
        cycleCount = 0;
    }
}

```

Neolight79, 2023-05-16:
What do you think concerning ```baroStartCalibration/baroSetGroundLevel```? They reset ```baroGroundAltitude``` thus during the calibration it will cause reset of accumulated data within ```performBaroCalibrationCycle```.
Maybe add condition in ```baroStartCalibration/baroSetGroundLevel```?

ledvinap, 2023-05-16:
@Neolight79 : IMO it would be better to make `cycleCount` file-static variable and reset it together with calibrationCycles ..

ledvinap, 2023-05-16:
The NaN (or Inf) is bad.
disabling `-ffast-math` is bad idea - it will reduce performance and current  code doesn't expect non-finite numbers anyway.

Maybe test cumulative error bits in FPCSR in main loop and indicate error to user. And allow raising exception for developers ...

Neolight79, 2023-05-16:
> @Neolight79 : IMO it would be better to make `cycleCount` file-static variable and reset it together with calibrationCycles ..

Thanks a lot! Agree, done.

Neolight79, 2023-05-16:
It works. But is this normal, such noise for altitude?
[image]

haslinghuis, 2023-05-16:
@Neolight79 are you going to PR this otherwise I'd like to add to #12815

Neolight79, 2023-05-16:
> @Neolight79 are you going to PR this otherwise I'd like to add to #12815

No problem, you can add it to #12815

haslinghuis, 2023-05-16:
> IMO it would be better to make cycleCount file-static variable and reset it together with calibrationCycles ..

@Neolight79 - see you are back to work - so take your time.
- Please check if something is missing in the PR.
- Where should we reset cycleCount ?
- Do we still need to move the DEBUG_SET in barometer.c ?
- Did you disable DEBUG_SET(DEBUG_BARO in position.c ?

Neolight79, 2023-05-17:
@haslinghuis unfortunately, work is not finished yet. I attached the working barometer.c, no changes to the other files. But the problem now is that altitude is not smooth. The pressure is changing in the same boundaries for DPS310, MS5611 in 4.4.1 of in 4.3.2, pressure (without moving the drone) changes +-5. Moreover, MS5611 is also smooth in 4.3.2, and DPS310 smooth in 4.4.1, thus I have to find the cause of such noisy altitude data. [barometer.zip](https://github.com/betaflight/betaflight/files/11495564/barometer.zip)

Neolight79, 2023-05-17:
I don't see any correlation between Pressure and Altitude, thus (I guess) pressure noise is not the casue:
[image]

Neolight79, 2023-05-17:
I have checked the spread of pressure values for DPS310 and MS5611. It seems that DPS310 has aproximately *4* and MS5611 aproximately  *11*. That means MS5611 is less precise but in 4.3.2 there is median filter which smoothing the noise. Thus, the problem with noise is wether in the MS5611 driver (maybe it is possible to get filtered data on HW level) or in the lack of additional smoothing in barometer.c

haslinghuis, 2023-05-17:
@Neolight79 updated the PR. :+1:

closed 2023-05-21 by haslinghuis

KoDaKrom, 2023-12-09:
Hi,
I have same issue with barometer, not recognized on BF 4.4.3
GPS GEP-M10-DQ
on
GEPRC Mark5 DC with AirUnit O3
FC+ESC SPAN F722-BT-HD V2
Magnetometer is ok (after firmware upgrade using custom cloud adding: Magnetometers + BARO_MS5611 BARO_DPS310

When I load configuration, something strange:
### ERROR IN resources: INVALID RESOURCE NAME: 'CAMERA_CONTROL'###
### ERROR IN set: INVALID NAME: baro_bustype = I2C###
### ERROR IN set: INVALID NAME: baro_i2c_device = 2###

I have not tried to downgrade to 4.3.2 because I would use the OSD HD functionality with my O3 air unit.

Some help or advice for resolution is welcome, thank you ;-)

haslinghuis, 2023-12-09:
When getting these errors the board does not include the `USE_BARO` define.

Additional have to add `BARO` to custom defines to fix for external baro device in this case.

KoDaKrom, 2023-12-10:
> When getting these errors the board does not include the `USE_BARO` define.
> 
> Additional have to add `BARO` to custom defines to fix for external baro device in this case.

PERFECT and THANK YOU !
Yes, it's working now.
I've added before flashing firmware (4.4.3) these custom adds:
_**BARO** BARO_MS5611 BARO_DPS310_
and **_Magnetometers._**
All external module are recognized (GPS, Magneto and Baro).

...I'm happy ;-)
Have a nice day

Seszi, 2024-03-05:
Hi Guys,
i cant get the external baro working in M-1025-MI. please help
I've added before flashing firmware (4.4.3) these custom adds:
BARO BARO_MS5611 BARO_DPS310
im not sure about the port settings, need to change anything there to make the baro available?
[image]

PR #12824 (2023-05-20): [4.4.2] Fix MS5611
Fixes: #12744

Backport of #12815