betaflight/betaflight #12746: Blackbox is missing for DYSF4PRO

pierre-quelin, 2023-04-30:
### Describe the bug

Hello,

It seems that the black box function has disappeared in the 4.4.1

T.Y.

### To Reproduce

Connect and go to tab Blackbox.
A message explain that the Blackbox is not present for my board.

### Expected behavior

A Blackbox must be present

### Support ID

```txt
2b2ddbb4-59b7-4cd6-9c73-5b379a91b2b4
```

### Flight controller

DYSF 4 PRO V2

haslinghuis, 2023-04-30:
Please flash with core mode enabled

See: http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

pierre-quelin, 2023-04-30:
It seems to work. T.Y.

This mode is disabled by default. I understood that it was a firmware without driver so why is it working ?
I tried to add "FLASH" in the other options but I couldn't find it.
I'm not sure how flashing works since version 4.4.0. My card is no longer auto-detected and the DFU mode must be forced.

haslinghuis, 2023-04-30:
Core mode is for support only. Please provide support ID using core build as we need the details.

pierre-quelin, 2023-04-30:
4cf405b0-5fac-4dc6-ab00-82c535d101c7

haslinghuis, 2023-04-30:
Thanks. Please enable show releases - and enable expert mode in firmware flasher tab.
Add `FLASH_M25P16` to custom defines.
For 4.5.0 you also need to add `FLASH` and flash again. If it works we can add this to configuration.

pierre-quelin, 2023-04-30:
I don't know if this was a problem. My configurator was version 10.8.0. Now, I have just updated it to 10.9.0.
What is "show releases" ? Release candidates ?
I use the 4.4.1 version,  I had "FLASH_M25P16" in custom definition and "SMARTPORT" telemetry.
There's no change. No Blackbox

haslinghuis, 2023-04-30:
Yes you need configurator 10.9.0 at least. Nightly build has fixed some issues and is backwards compatible.

Ok try `FLASH_W25Q128FV` 
If that does not work try `FLASH FLASH_W25Q128FV` as it is different in 4.4.0 / 4.4.1 / 4.5.0

pierre-quelin, 2023-04-30:
FLASH FLASH_W25Q128FV is working fine with the 4.4.1 version.
Thank you very much.
Do you need another test ?

haslinghuis, 2023-04-30:
Thanks. We have all info we need.

pierre-quelin, 2023-04-30:
Thank you very much

closed 2023-04-30 by pierre-quelin