betaflight/betaflight #12750: Problem with the wrong MPU

SirBarlin, 2023-05-01:
### Describe the bug

I have a problem with the FC HGLRC ZEUS10 AIO 2-6S F411 Flight Controller ESC there is no gyro and no acc displayed I have already tried to flash another firmware on it but also does not work 

### To Reproduce

It seems that HGLRC has installed a different MPU than specified.

### Expected behavior

Please solve the problem Thank you

### Support ID

```txt
ID: "5e1352bf-8773-4be6-a08e-2ea9f879f8ef"
```

### Flight controller

HGLRC ZEUS10 AIO 2-6S F411 Flight Controller ESC

SirBarlin, 2023-05-01:
What do you mean by the link I did not quite understand sorry?

haslinghuis, 2023-05-01:
Please follow the 3 steps in the link to provide support ID from a core only build:

[image]

[image]

SirBarlin, 2023-05-01:
ID: "526e00f7-9bb2-4ed0-95fd-29fb7be7c042"

haslinghuis, 2023-05-01:
There seems to be a problem with this target with 4.4.1
Please try:
```
set gyro_2_spibus = 1
save
```

If that does not help you can try 4.5.0

SirBarlin, 2023-05-01:
what do you mean by version 4.5.0 that is not yet available or? the first version did not work.

SirBarlin, 2023-05-01:
or do I have to load the firmware again with ACCGYRO_BMI270 in the user definitions?

haslinghuis, 2023-05-01:
BMI270 has been defined for the board. There is a configuration issue with some boards on 4.4.1 and it looks like this is one of them. Setting `gyro_2_spibus = 0` did not work? Until we can find a fix please either keep using 4.4.0 or try 4.5.0.

haslinghuis, 2023-05-01:
[image]

SirBarlin, 2023-05-01:
with version 4.4.0 it does not work either and 4.5.0 is not available yet

haslinghuis, 2023-05-01:
Change release to development to make 4.5.0 available. We still need a core only build to reveal hardware available.

haslinghuis, 2023-05-01:
According to the manufacturer product page we need target `ZEUSF4EVO`

https://www.hglrc.com/products/hglrc-zeus10-aio-fpv-racing-drone-2-6s-f411-flight-controller-10a-bl_s-4in1-esc

SirBarlin, 2023-05-01:
what do you mean by target? 4.5.0 does not work either

haslinghuis, 2023-05-01:
Sorry for all terminology - each flight controller has a target name for the board needed for configuration.

Looks like you will need to choose `ZEUSF4EVO` instead of `HGLRCF411`

SirBarlin, 2023-05-01:
with the Zeusf4EVO it also does not work. Do not wonder that I now no longer anworte I have to work tomorrow

SirBarlin, 2023-05-01:
with the Zeusf4EVO it also does not work. Do not wonder that I do not answer now I have to work tomorrow

haslinghuis, 2023-05-02:
- Did you try core only build with ZEUSF4EVO on 4.4.0 / 4.4.1 / 4.5.0?
- Verify the correct target with the reseller?

SirBarlin, 2023-05-09:
Hello,

I'll try this again at the weekend in peace I just have a lot of stress at work

SirBarlin, 2023-05-14:
Hello,

I have tried all versions, but the problem is still there, I have always made sure that it is in the core build. Do you have any ideas?

haslinghuis, 2023-05-14:
Try core build on:

[image]

closed 2023-06-01 by haslinghuis