betaflight/betaflight #12751: SD Card not working in 4.4.1 SpeedyBee F405 V3, works fine in 4.4.0

pagex, 2023-05-01:
### Describe the bug

Going to Blackbox it says Fatal error, reboot to retry
Re-flashing immediately with 4.4.0 detects SD card fine.

Noticed there is no recourse SDCARD_CS inside resource command see below.

### To Reproduce

Flash with 4.4.1
Apply Defaults
Go to Blackbox

### Expected behavior

Blackbox detects card

### Support ID

```txt
version
# Betaflight / STM32F405 (S405) 4.4.1 Apr 10 2023 / 21:32:29 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF405V3

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=42degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3641, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: b9cd90a43a78b121e960fac45c6d7124 (4.4.1)
System Uptime: 19 seconds, Current Time: 2023-05-01T11:46:48.234+00:00
CPU:14%, cycle time: 317, GYRO rate: 3154, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.1 Apr 10 2023 / 21:32:29 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF405V3
manufacturer_id SPBE

# resources
resource BEEPER 1 C05
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B05
resource MOTOR 8 B04
resource LED_STRIP 1 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
[... 389 lines cut ...]
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#

haslinghuis, 2023-05-01:
Fixed in: https://github.com/betaflight/betaflight/pull/12671

Squizzy911, 2023-05-02:
> Fixed in: https://github.com/betaflight/betaflight/pull/12671

Yep but not for standard user because no release yet :( is there an ETA after milestones are finished and no open PR?

StNekroman, 2023-05-22:
In 4.4.1 not working at all.
In 4.4.0 working if sdcard is formatted using 3rd party tool.
Using Windows built-in format -  not work.

closed 2023-05-22 by haslinghuis

dukeflipchart, 2023-06-16:
I'm running Betaflight 4.4.2, and I'm having a similar issue - SD card is detected by windows just fine, but Betaflight says fatal error. Speedybee F405 V3 FC.

Tried reformatting card with Windows and also with SD Memory Card Formatter, none worked.

Any tips?

haslinghuis, 2023-06-16:
@dukeflipchart this PR was about configuration.

There is an issue with shared SPI:  https://github.com/betaflight/betaflight/issues/12583