betaflight/betaflight #12756: Flight Controller F7 freezes when JETIEXBUS is used on Betaflight 4.4

FlyMarc2, 2023-05-03:
### Describe the bug

Betaflight 4.4.0 and 4.4.1 is not reliable using JETIEXBUS protocol. Using this protocol the telemetry and FC randomly and suddenly freezes. The motors shut down and the drone crashes. The transmitter no longer receives telemetry data from the FC and the drone no longer responds to any commands (beeper, arm/disarm, etc.). The OSD display of the goggles also no longer changes. The receiver telemetry still works (RX BEC Voltage is still displayed and logged on transmitter. Only unplugging and plugging in the lipo brings the drone back to life.
Blackbox recording also stops when FC freezes.
Another thing I noticed is, VARIO Data is not transmitted anymore. Neither to the OSD nor to the transmitter.
It seems to be the same problem as with iNav (https://github.com/iNavFlight/inav/issues/8899).
Everything worked fine with 4.2.11 and 4.3.2. When I go back to firmware 4.2.11 or 4.3.2 everything works again. Exactly the same hardware and wiring.

### To Reproduce

It happens every 2-3 flights. Mostly when armed once. Also on workbench without flying.

### Expected behavior

Firmware 4.4 should be as reliable as 4.2 and 4.3 with JETIEXBUS protocol. Never had an issue with JETIEXBUS before 4.4

### Support ID

```txt
Id: "bf18af1d-9918-48d2-b6fc-ec6022077788"
```

### Flight controller

iFlight BLITZ Mini F722 with BLITZ Mini E55S 4-IN-1 ESC

### Other components

Transmitter: Jeti DS 16 II FW: 5.06
Receiver: Jeti REX 3 (set on EXBUS) FW: 1.15
FC: BLITZ Mini F722 FW:?
ESC: BLITZ Mini E55S 55A 4-IN-1 ESC FW: BLHeliS 16.7 or Bluejay 0.16, 24kHz
Video Transmission: SucceX Mini 600mW VTX FW:?
Motor: XING-E Pro 2207  motors
GPS: iFlight M8Q-5883 GPS Modul und Kompass V2.0
Drone: BNF iFlight Nazgul Evoque F5 6S (analog)

### How are the different components wired up (including port information)

VTX: UART1
Receiver: UART2
GPS: UART4
ESC: UART6
[btfl_003.bbl.zip](https://github.com/betaflight/betaflight/files/11380795/btfl_003.bbl.zip)

haslinghuis, 2023-05-03:
Try: [IFLIGHT_BLITZ_F722.zip](https://github.com/betaflight/betaflight/files/11390972/IFLIGHT_BLITZ_F722.zip)
(_Betaflight 4.4 maintenance compiled with ARM9_)

Duplicate of https://github.com/betaflight/betaflight/issues/12140

Also please try 4.5.0

FlyMarc2, 2023-05-04:
Thanks a lot. Version above now works perfectly. No freezes of the FC anymore! Telemetry and vario are also working. Vario only works when armed and altitude changes to „0“ when armed (ok) and changes back to GPS altitude when disarmed (not ok). I think in 4.3.2 vario also worked when disarmed and altitude showed „0“ when armed, but did not change to GPS altitude when disarmed. I found that better. Is it possible to make it work like in 4.3.2?

If I try 4.5.0, can I use the presets of 4.4? Especially the PID tunings, because now it flies perfectly in 4.4

haslinghuis, 2023-05-04:
Presets for 4.5.0 will be made available after of during release as we might have breaking changes.
But if you save a diff most settings will be compatible :crossed_fingers: 

It's really odd that using previous GCC solves this problem for JETI as this is afaik the only hardware with the freezing issue.

robertrullyp, 2023-08-07:
Got the same issue here with FC BetaFPV F722 AIO 35A BLHeli_32, shall i open for new issue?
oh fi, At that time, my goggles were still receiving live video, but all OSD parameters were frozen.

haslinghuis, 2023-08-09:
@robertrullyp this issue is related to JETIEXBUS. Please try 4.4.3-zulu or 4.5-zulu builds.

fabi9163, 2023-08-18:
@FlyMarc2 I've the same issue, discussed in another thread. Did you know how you flashed your firmware on which the problem appeared? With "erase flash" or did you reflash a minor version without erase the whole chip?

FlyMarc2, 2023-08-19:
@fabi9163 
I made it this way:
Download the file [IFLIGHT_BLITZ_F722.zip from haslinghuis:(https://github.com/betaflight/betaflight/files/11390972/IFLIGHT_BLITZ_F722.zip)
Then open Betaflight, select update Firmware, Auto-detect, Full chip erase, Radioprotocol JETIEXBUS, Other options, Telemetry protocol and Motor protocol as shown.
Then press pleas load firmware file (the downloaded zip file) and flash firmware.
Then you have to reconfigure Betaflight. Before I made screenshots of each menu.
Works fine since I flashes this firmware. Not one issue and crash anymore!!! Telemetry also works fine.

[image]

haslinghuis, 2023-08-19:
Local firmware has the JETIEXBUS and VARIO included by default. No need for cloud configuration as you are flashing a local firmware.

Happy to create 4.4.3 test versions when you provide a build key (which contains the build command cloud build uses).

klutvott123, 2023-09-16:
are there any leds blinking when this happens?

RoadyFPV, 2023-09-17:
no is totaly freezed only red led is lit up permanently

RoadyFPV, 2023-09-17:
https://github.com/betaflight/betaflight/issues/12140

sugaarK, 2023-09-19:
> If I try 4.5.0, can I use the presets of 4.4? Especially the PID tunings, because now it flies perfectly in 4.4

Tuning wise you’ll be fine

haslinghuis, 2023-09-24:
We are not going to downgrade ARM SDK before proper investigation. Having requested samples from JETI but got no response at all.

klutvott123, 2023-10-21:
Can you guys please try this https://github.com/betaflight/betaflight/pull/13130

closed 2023-10-21 by SteveCEvans

RoadyFPV, 2023-11-12:
MatekF722se freeze after a couple of minutes

fabi9163, 2023-11-12:
Hi @RoadyFPV, please share some more details. On which version did this happen to you, etc?

RoadyFPV, 2023-11-12:
MATEKF722SE
4.5 zulu
GPS
JETIEXBUS
DJIFPV

klutvott123, 2023-11-12:
Well that sucks. Exactly the same behaviour as before?

RoadyFPV, 2023-11-12:
yes i think so, but mambaf722_i2c dont freeze that confuses me

klutvott123, 2023-11-12:
@RoadyFPV All right. So it works without telemetry, but freezes when it's enabled? Must be more to this then. 
Please triplecheck that the matek is actually on the right firmware with the fix

RoadyFPV, 2023-11-12:
@klutvott123  OK i will check that tomorrow,

yea test was a fresh flash with 4.5 zulu.

I will test:
1 flash again with 4.5 zulu
2 check with telemetry option enabled and disabled

RoadyFPV, 2023-11-12:
What i can say if i turn off transmitter, it runs for 4 hours yesterday

klutvott123, 2023-11-12:
@RoadyFPV Maybe it's because telemetry requests come from the transmitter. Another thing is that it would be in failsafe and the framerate is reduced. check RX task in cli with transmitter off to see the framerate.

RoadyFPV, 2023-11-12:
ok to check RX in cli i type "tasks"?

klutvott123, 2023-11-12:
@RoadyFPV Yes, that's where you check it. Failsafe off in receiver probably just means that it doesn't do anything on failsafe. Betaflight should detect it anyway on failsafe flags and apply the BF failsafe behaviour.

RoadyFPV, 2023-11-13:
Here an task with TX on

[image]

RoadyFPV, 2023-11-13:
@klutvott123  were you able to reproduce the freezing of the fc with the pyton script?

RoadyFPV, 2023-11-13:
i have now activatet one after the other.

Telemetry disabled, UART 3 GPS no freeze
Telemetry disabled, UART 3 GPS, UART 6 SMARTAUDIO - instant freeze
Telemetry disabled UART 3 disabled, UART 6 SMARTAUDIO - no freeze until now

RoadyFPV, 2023-11-13:
As far as i us UART 1, 3, 6 Smart Audio it freezes in under 1 min

I have now uart 1, 3, 6 MSP+Displayport with Telemetry enabled - Test is running right now

elhsmart, 2024-08-17:
@RoadyFPV sorry for necroposting, did you figure out your MSP freeze? Stuck with same behaviour recently

RoadyFPV, 2024-08-18:
@elhsmart only switch to h743 can solve my issue here. I think it's due to limited memory on f7. i cant notice any frreeze on the h743. neither on bf nor on inav i recommend using it with h743 fc's