betaflight/betaflight #12766: mamba f405 mk2 no baro 4.4

skinnyguyfpv, 2023-05-05:
### Describe the bug

hi chaps, ive watched Joshua's videos and also did a support data id on discord but still cannot get it sorted out, i am using an external BMP280 baro which worked fine in BF 4.3, please help

### To Reproduce

flash 4.4, select barometer, save and reboot ikon disappears 

### Expected behavior

save and reboot, have barometer working

### Support ID

```txt
Id: "fb01cb18-af4a-4725-bff0-5e016379ac6c"
```

### Flight controller

mamba f405 mk2

### Other components

crsf, rush racing vtx, BF 4.4

skinnyguyfpv, 2023-05-05:
I'm still really new to this, have you created a commit? If so what number is it? Thankyou

haslinghuis, 2023-05-05:
https://github.com/betaflight/unified-targets/pull/1001

skinnyguyfpv, 2023-05-05:
> [betaflight/unified-targets#1001](https://github.com/betaflight/unified-targets/pull/1001)

Just tried #12766 with no luck

closed 2023-05-05 by haslinghuis

skinnyguyfpv, 2023-05-05:
> [betaflight/unified-targets#1001](https://github.com/betaflight/unified-targets/pull/1001)
> 
> In the mean time just add `BARO_BMP280` (or `BARO BARO_BMP280`) to custom defines when building firmware.

Just tried this aswell, no luck

skinnyguyfpv, 2023-05-05:
Just tried BARO_BMP280 also with no luck

skinnyguyfpv, 2023-05-05:
Ok don't know what I did differently but it's working 👍