betaflight/betaflight #12768: [CONFIG] SynergyF4 missing Flash

nerdCopter, 2023-05-06:
### Describe the bug

* support-id https://build.betaflight.com/api/support/ce16576f-6d67-4867-a33d-c3603993d754

`SYNERGYF4`
`4.4.1`
`FLASH_W25Q128FV` according to log

### To Reproduce

flash 4.4.1

### Expected behavior

flash pre-defined

### Support ID

```txt
ce16576f-6d67-4867-a33d-c3603993d754
```

### Flight controller

SYNERGYF4

haslinghuis, 2023-05-06:
https://github.com/betaflight/config/blob/master/configs/KLEE-SYNERGYF4/config.h

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis