betaflight/betaflight #12770: NO DATAFLASH CHIP FOUND (CANT ACCES BLACKBOX)

Swordfishhh, 2023-05-07:
### Describe the bug

Flash FC memory dont recognize by BF 4.4.1
FC IS HAKRCF722D

### To Reproduce

CONNECT HAKRCF722D TO BF AND CHECK THE BLACKBOX, THERE IS ONLY "NO LOGGING" AND "SERIAL PORT" NO ONBOARD BLACKBOX OPTION

### Expected behavior

CONNECT TO INTERNAL FC BLACKBOX

### Support ID

```txt
3a7d425d-99e2-40ee-8121-5d97056d1c5a
```

### Flight controller

HAKRCF722D

### Other components

NONE

### How are the different components wired up (including port information)

NONE

Swordfishhh, 2023-05-07:
> Try 4.4.0 or 4.5.0

I did try all firmwars and didnt work

haslinghuis, 2023-05-07:
Follow these steps: https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

closed 2023-06-01 by haslinghuis