betaflight/betaflight #12771: Barometer not detected while flashing 4.4.1

Andrea3v, 2023-05-07:
### Describe the bug

Just want to report the missing drivers for the gyro/accelerometer of my Eachine F4 flight controller (on Eachine Tyro 119).
added them 'manually':

1. firmware mode > flashed "core only"
2. [CLI > submit support data, see ID below]
3. found the correct drivers (GYRO=MPU6000 and ACC=MPU6000) at the top of the CLI (status)
4. https://betaflight.com/docs/manufacturer/manufacturer-design-guidelines and got the corresponding defines for the gyro/acc
5. back into flash firmware mode > show release canditate + expert mode
6. added "GYRO_SPI_MPU6000" and "ACC_SPI_MPU6000"  under 'custom define'
7. load firmware online + flash 

### To Reproduce

flash Betaflight 4.4.1 onto Eachine GPS F4 MATEK405STD FC.

### Expected behavior

flash the correct drivers (GYRO_SPI_MPU6000 and ACC_SPI_MPU6000) for the GPS F4

### Support ID

```txt
b7e6f460-c014-441f-b71a-ea0b6232cc6f
```

### Flight controller

Eachine GPS F4 MATEKF405

jsk2084, 2023-05-07:
Hi! I have this FC also, and learned that the matekf405std_clone is the correct target for it. Everything should work when flashed using that.

haslinghuis, 2023-05-07:
Confirm. I have created the clone target to mitigate from drivers missing in the OEM target.

closed 2023-05-07 by haslinghuis