betaflight/betaflight #12774: GYRO ICM-42688P on AXISFLYINGF7PRO

AlGix, 2023-05-07:
### Describe the bug

Hi,

I have a BNF Axisflying Manta 5 DC drone equipped with the ARGUS PRO 55A stack.
In Betaflight, this FC is provided with a BMI270 (same to manufacturer site), but unfortunately it has an ICM-42688P.
In order for Betaflight to work, I had to create a specific build with the right arguments, because otherwise i'm in "no gyro".
Can you add this gyro to this flight controller?

The gyro also makes a lot of noise. The 4.4.1 update considerably improved this, but it's still not great.
I join some blackboxes. Can you work on this gyro? Do you have any filter tips to improve my drone?

Finally, when compiling the build, the "FLASH" option is not available on this FC while the BlackBox is working.

Last observation, I always have 5 I2C errors and I have no idea what that corresponds to?

### To Reproduce

Blackbox with FC ICM42688-P

### Expected behavior

Include ICM-42688P on AXISFLYINGF7PRO
Less noise on the ICM-42688-P
Enable FLASH option on AXISFLYINGF7PRO
No I2C with this FC

### Support ID

```txt
8472063-b2aa-407a-ba16-430b932778d3
```

### Flight controller

AXISFLYINGF7PRO

### Other components

BETAFLIGHT 4.4.1
GYRO - ACC : ICM42688-P
STACK : AxisFlying ARGUS PRO 55A

haslinghuis, 2023-05-07:
The support ID does not seem to be valid.

http://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

AlGix, 2023-05-07:
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr  6 2023 / 21:22:54 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: AXFL, board_name: AXISFLYINGF7PRO

# status
MCU F722 Clock=216MHz, Vref=3.21V, Core temp=41degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3271, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MSP (53 x 20)
BUILD KEY: 4c687b31e6cb462b1f8c2cc0e442e431 (4.4.1)
System Uptime: 1072 seconds, Current Time: 2023-05-07T19:28:17.551+00:00
CPU:53%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 5
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP RPMFILTER

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=16777216

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.1 Apr  6 2023 / 21:22:54 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name AXISFLYINGF7PRO
manufacturer_id AXFL

# resources
resource BEEPER 1 C13
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B10
resource MOTOR 8 B11
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
[... 217 lines cut ...]
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Données de support soumises *** 
ID : "9a27c5b4-8f65-4efc-b307-6826a4923296"

# Copier l'ID et le fournir à l'équipe de betaflight

AlGix, 2023-05-07:
[image]

AlGix, 2023-05-07:
[image]

haslinghuis, 2023-05-07:
Fixed in 4.5 also should work in 4.4.0

AlGix, 2023-05-07:
U have link for this lot patch ?

haslinghuis, 2023-05-07:
Add `GYRO_SPI_ICM42688P ACC_SPI_ICM42688P` in custom defines before flashing (enable show release candidates and expert mode)

AlGix, 2023-05-07:
Man, I think you just didn't read my questions.
I've used the custom build before and it works. My requests are quite different.

haslinghuis, 2023-05-07:
My message was in reply to:

> In order for Betaflight to work, I had to create a specific build with the right arguments, because otherwise i'm in "no gyro".
Can you add this gyro to this flight controller?

These defines are missing in https://github.com/betaflight/betaflight/blob/4.4-maintenance/src/config/AXISFLYINGF7PRO/config.h

Not addressed:

> Finally, when compiling the build, the "FLASH" option is not available on this FC while the BlackBox is working.

Define the FLASH option is not available

> Last observation, I always have 5 I2C errors and I have no idea what that corresponds to?

Baro is using I2C

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis