betaflight/betaflight #12780: BF4.4.1 Applying custom defaults on F722 flight controllers bricks them (need to reflash in DFU mode)

ChrisRosser, 2023-05-09:
### Describe the bug

BF4.4.1 when flashed to F722 targets (IFLIGHT F722 TWING and IFLIGHT F722 BLITZ) works but when custom defaults are applied the device will no longer connect over USB (device descriptor request failed).

Issue not present in BF4.4.0

### To Reproduce

Flash BF4.4.1 to F722 flight controller and apply custom defaults

### Expected behavior

Should work

### Support ID

```txt
40e387b8-cf29-4741-b20b-c2332a2632fe
```

### Flight controller

IFLIGHT F722 TWING

haslinghuis, 2023-05-09:
Duplicate of https://github.com/betaflight/betaflight/issues/12635

@ChrisRosser please test [betaflight_4.4.2_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11431734/betaflight_4.4.2_STM32F7X2.zip)

Berlinologe86, 2023-05-14:
> Duplicate of #12635
> 
> @ChrisRosser please test [betaflight_4.4.2_STM32F7X2.zip](https://github.com/betaflight/betaflight/files/11431734/betaflight_4.4.2_STM32F7X2.zip)

hello I just tested it on the IFLIGHT F722 TWING mini with the 4.4.2. After setting the defaults, I can access FC but have no configuration. no gyro frequency, accelerometer, uarts, nothing
 dump all:

# 

# dump all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 May  9 2023 / 15:44:12 (d335dafb2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC
mcu_id 003100513338510b39393832
signature 

# name: -

# resources
resource BEEPER 1 NONE
resource MOTOR 1 NONE
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 NONE
[... 1297 lines cut ...]
set rates_type = ACTUAL
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

haslinghuis, 2023-05-15:
Please try 4.4.2 and 4.5.0 dev build. Also please report support ID instead of showing a dump.

Steow, 2023-05-16:
> Please try 4.4.2 and 4.5.0 dev build. Also please report support ID instead of showing a dump.
> 
> [image]

Hello, I have the same problem than Criss Rosser with a Iflight F722 Twing Mini with BF4.4.1. Both 4.4.2 and 4.5.0 seems to work well after flash, also I did not fly them. Thanks for the work !