betaflight/betaflight #12790: RX MSP OVERRIDE: Using MSP_SET_RAW_RC together with another serial RX

cvetaevvitaliy, 2023-05-11:
### Is your feature request related to a problem? Please describe

I can only enable RX_MSP or RX_SERIAL, not both. I want to have a single board computer running on the drone (and I like Betaflight 1st_place_medal, so I don't want to use other firmware such as Ardupilot where through Mavlink I can control the drone) for autonomous flight 

### Describe the solution you'd like

Add a feature to the `Modes` or `Adjustments` tabs that will allow us to flexibly adjust the mode: RX MSP OVERRIDE
In detail:
* Add option to configurator, select any RC channel (except of course Roll, Pitch, Yaw, Thr) to enable 'RX MSP OVERRIDE' feature
* After activating this mode 'RX MSP OVERRIDE', only the Roll, Pitch, Yaw, Thr channels should be overwritten from MSP, all others from the receiver.
* FailSafe should be unaffected and depend on the RC receiver (If the connection is lost - need to activate FS)

This will allow us to activate this mode 'RX MSP OVERRIDE' at the command of the remote control, as well as to turn it off at any time if the "autonomous flight" is out of control

### Describe alternatives you've considered

I tried an autonomous flight on Ardupilot, there it is possible to send RC commands using the Mavlink protocol, but I don't like how it flies, Betaflight is best. Betaflight has the Mavlink protocol but only sends telemetry data.

haslinghuis, 2023-05-12:
Any info on the SBC and software used?

> I can only enable RX_MSP or RX_SERIAL, not both. I want to have a single board computer running on the drone (and I like Betaflight 1st_place_medal, so I don't want to use other firmware such as Ardupilot where through Mavlink I can control the drone) for autonomous flight

cvetaevvitaliy, 2023-05-13:
the best software! 😀 

The NPU to quickly run neural network and many thousands of code

haslinghuis, 2023-05-13:
Any repo available on that ?

cvetaevvitaliy, 2023-05-13:
For the part that communicates with the flight controller via MSP, will definitely be published, I have already started working on it.

sending RC commands via MSP works, i want to add other ʼmsp commandsʼ to get telemetry and stuff...

haslinghuis, 2023-05-14:
https://github.com/betaflight/betaflight/issues/8292

cvetaevvitaliy, 2023-05-14:
> https://github.com/betaflight/betaflight/issues/8292

I know about this issue. I read everything, but the RX architecture in Betaflight is made so that only one receiver can work. 
I suggest providing the possibility to activate this RC MSP overwriting mode from the remote control and also turn it off.

About which NPU, there are already many of them built into CPUs from different manufacturers. 

I'm currently working with Rockchip, they have different performances NPU, from 1 tops to 6 tops.

Right now I have a small prototype the size of a flight controller.
this is a render so far, PCB prototype fabrication in progress.

[image]

[image]

[image]

[image]

haslinghuis, 2023-05-14:
Thanks. Interesting. Looking good :heart_eyes: 
Does it run opencv ?

Would this PR not allow what you are asking for: https://github.com/betaflight/betaflight/pull/9915 ?
Would fixing MAVlink be the answer?

cvetaevvitaliy, 2023-05-15:
> 

No. This PR does`t resolve this feature, always RC commands will be rewritten form receiver.
https://github.com/betaflight/betaflight/blob/master/src/main/rx/rx.c#L654-L662
You can test it, if the receiver is selected any other than MSP, the MSP_SET_RAW_RC commands will be ignored

To be able to overwrite commands with MSP_RC, need to rewrite this code something like this
the FailSafe functionality will not be affected in any way.

```C
        if (msp_ovveride_activate && channel < NON_AUX_CHANNEL_COUNT)
            sample = rxMspReadRawRC(&rxRuntimeState, rawChannel);
        else 
            sample = rxRuntimeState.rcReadRawFn(&rxRuntimeState, rawChannel); 
```

where the flag `msp_ovveride_activate` can be activated/deactivated from the `Modes` or `Adjustments `
Also will be overwritten only: Roll, Pitch, Yaw, Thr channels

In this way, we will be able to flexibly configure this feature

cvetaevvitaliy, 2023-05-23:
@haslinghuis 
So, do you want to add feature to the Betaflight configurator, if I implement the functionality on the Betaflight side
I mean, the ability to flexibly configure which RC channel will be responsible for enabling/disabling the listening of RC NON_AUX_CHANNEL commands from the MPS

haslinghuis, 2023-05-23:
Yes like to assist in client implementation. Are you able to provide a sample for testing?

cvetaevvitaliy, 2023-05-26:
> 

Yes, I can provide sample for testing

haslinghuis, 2023-05-26:
You can contact me at my name at firmware name dot com

cvetaevvitaliy, 2023-05-31:
@haslinghuis 
Hi!
I added a new mode, now you can configure any free AUX channel
After its activation, four control channels: Roll, Pitch, Yaw, Thr will be listened to from the MSP. After deactivation, control of the "sticks" will return to the pilot

Thus we have the opportunity to use MSP_SET_RAW_RC command together with another(any) RX, without breaking Failsafe mode and without overwriting other AUX channels

danyayoe, 2023-05-31:
@cvetaevvitaliy so, you've added a possibility to control a drone with betaflight firmware via mavlink protocol? Am I right? If yes, could you please give more info or a link how to configure it?

cvetaevvitaliy, 2023-05-31:
> 

No, via MSP protocol, BetaFlight supports mavlink but only sending telemetry data, without listening RC commands

danyayoe, 2023-05-31:
Got it, thanks for the answer anyway. So you use MSP protocol to control the Betaflight drone from the single board computer?

danyayoe, 2023-05-31:
@cvetaevvitaliy with pleasure

ledvinap, 2023-06-01:
@cvetaevvitaliy : Using RC override is good for initial testing, but eventually, it would be better to control PID setpoints directly. There was some PR with quaternion as angle mode setpoint

cvetaevvitaliy, 2023-06-01:
> 

Thanks for reply. Tell us in detail about PID setpoints, how it can be controlled by MSP

tgalkovskyi, 2023-07-17:
I was able to verify that with betaflight master:

1. having firmware with msp_override_channels_mask fix https://github.com/betaflight/betaflight/pull/12857
2. and setting msp_override_channels_mask=0xf in betaflight cli

was enough for me to be able to issue MSP_SET_RAW_RC over either USB or non-usb UART port in MSP_OVERRIDE mode, no other changes were necessary.
MSP_OVERRIDE is bound to AUX3 and I can go into and out of this mode by switching AUX3, where my MSP_SET_RAW_RC are only respected in MSP_OVERRIDE mode and ignored otherwise.

haslinghuis, 2023-07-17:
Fixed in #12857

closed 2023-07-17 by haslinghuis

Automate5462, 2023-12-10:
> Hi! I added a new mode, now you can configure any free AUX channel After its activation, four control channels: Roll, Pitch, Yaw, Thr will be listened to from the MSP. After deactivation, control of [...]

@cvetaevvitaliy Thanks for adding this mode!
I tried to achieve the same, but it did not work:
When msp_override_channels_mask is set to zero in betaflight cli, then the msp override mode disappears.
And when it is set to e.g. =1, then the msp override mode can be set, and assigned to e.g. AUX2.
But once the AUX2 triggers this mode, it is not possible to exit from this mode, as the AUX channels are not responding anymore to the transmitter.
Do you know, what needs to be done, so that the AUX-channels are not overwritten by the msp, but always taken from the transmitter?
(So that the transmitter can always start and stop the mode, when the msp overwrites the pitch, roll, yaw and throttle?)

cvetaevvitaliy, 2023-12-10:
> @cvetaevvitaliy Thanks for adding this mode! I tried to achieve the same, but it did not work: When msp_override_channels_mask is set to zero in betaflight cli, then the msp override mode [...]

Hi! Thanks for your interest in my idea, but my code was not merged )

Automate5462, 2023-12-10:
@cvetaevvitaliy Thanks a lot for your fast reply. 
Do you still have the files / are they still somewhere on github?
So that I could build the firmware and flash this custom version onto an STM32F405 FC as a target?

cvetaevvitaliy, 2023-12-10:
> @cvetaevvitaliy Thanks a lot for your fast reply. Do you still have the files / are they still somewhere on github? So that I could build the firmware and flash this custom version onto an STM32F405 FC as a target?

I don't have it anymore, I work immediately with angles in PID controller

dronieee, 2024-01-06:
> > @cvetaevvitaliy Thanks a lot for your fast reply. Do you still have the files / are they still somewhere on github? So that I could build the firmware and flash this custom version onto an STM32F405 FC as a target?
> 
> I don't have it anymore, I work immediately with angles in PID controller

@cvetaevvitaliy Hi, I have exactly the same use-case you described in the first post. How do you control the angles from the single board computer using MSP/SERIAL? Could you please point me to any resources that would help me achieve this? Thanks!

devnim, 2024-01-14:
> > 
> That's right! 
> I see that you are also from Ukraine, if we have the same purpose, you can contact me
> 

How do I contact you?

cvetaevvitaliy, 2024-01-15:
@devnim 
You can write in BF Discord

devnim, 2024-01-15:
@cvetaevvitaliy, I cannot find you on BF discord. Are you using a different username there?

cvetaevvitaliy, 2024-01-15:
@devnim 
See hardware topic

devnim, 2024-01-15:
@cvetaevvitaliy, I do not know what I am doing wrong, but I do not see this topic in BF discord. Could you please add me? Either through this link  https://discord.com/invite/qj4SVu4z

Or by my discord username pal0679_27641

devnim, 2024-01-21:
Hi @Automate5462 , did you find a solution to the problem you encountered below? 

> I tried to achieve the same, but it did not work:
> When msp_override_channels_mask is set to zero in betaflight cli, then the msp override mode disappears.
> And when it is set to e.g. =1, then the msp [...]

devnim, 2024-01-21:
@dronieee , did you find a way or a resource on how to do it?

> 
> @cvetaevvitaliy Hi, I have exactly the same use-case you described in the first post. How do you control the angles from the single board computer using MSP/SERIAL? Could you please point me to any resources that would help me achieve this? Thanks!

Automate5462, 2024-01-22:
No, I have not found the solution.
> Hi @Automate5462 , did you find a solution to the problem you encountered below? 
> 
...

sasha-vtz, 2024-07-23:
@cvetaevvitaliy, how can we contact?
I'm from Ukraine, I share the same purpose as every Ukrainian)
I want to configure MC overwrite on my FPV drone to be able to control it via Raspberry PI

cvetaevvitaliy, 2024-07-23:
> @cvetaevvitaliy, how can we contact?
> 
> I'm from Ukraine, I share the same purpose as every Ukrainian)
> 
> I want to configure MC overwrite on my FPV drone to be able to control it via Raspberry PI

Vision...

sasha-vtz, 2024-07-23:
@cvetaevvitaliy, I have flashed my FC SpeedyBee with the 4.5 firmware, configurator 10.10
But I do not see the MSP OVERRIDE Mode among the modes list
Is that hidden?
I think its better if we communicate about that somewhere else
My Insta is sasha.vtz
I think I have found you on instagram

cvetaevvitaliy, 2024-07-23:
> @cvetaevvitaliy, I have flashed my FC SpeedyBee with the 4.5 firmware, configurator 10.10
> 
> But I do not see the MSP OVERRIDE Mode among the modes list
> 
> Is that hidden?
> 
> I think its better if we communicate about that somewhere else
> 
> My Insta is sasha.vtz
> 
> I think I have found you on instagram

This feature is not included in any release

under0tech, 2025-02-24:
MSP_OVERRIDE is available in the 4.6 dev version of Betaflight, but you need to properly configure it.
I have created a comprehensive guide on how to configure and use this feature with Python: https://github.com/under0tech/autopilot_bee_ept

and an appropriate article on Medium (https://medium.com/@dmytrosazonov/fpv-autonomous-operation-with-betaflight-and-raspberry-pi-0caeb4b3ca69) with a detailed explanation.

I believe that after reading these materials, you will not have any more questions on this topic.

redspry, 2025-03-03:
if you use betaflight 4.6 note the configurator 10.10 most recent release is deprecated -- and you need to build the betaflight-configurator repo from source to connect to and flash your controller -  anyways same issue --- working with betaflight as part of a competition for a2rl and the autonomy overide default features are horribly not user friendly - an open source user friendly release for autonomy would be ideal. I am using the christian raunch multiwii project and trying to modify. 

with RX_MSP still having trouble successfully arming via betaflight to allow flight

Major request to open an autonomy specific channel in the betaflight discord ~