betaflight/betaflight #12794: Hardware vault after lastest master

TheIsotopes, 2023-05-11:
### Describe the bug

since latest master https://github.com/betaflight/betaflight/commit/f44cc9fa6e371baa0d92bc91acadb0a7218dcabe the cli command `set dshot_bidir = ON` makes a hardware vault on MATEKF722MINI

@ctzsnooze pls can you check this in #12688

### To Reproduce

use latest master and the cli command `set dshot_bidir = ON` and save

### Expected behavior

 working bidir dshot on my fc

### Support ID

```txt
Unable to create support id due to hardware vault.
```

### Flight controller

MATEKF722MINI

haslinghuis, 2023-05-11:
Matek H743 is ok.
Speedybee F7 hard-fault after saving the setting.

haslinghuis, 2023-05-11:
@TheIsotopes please test `#12796` (paste or type into commit field)

TheIsotopes, 2023-05-12:
@haslinghuis it's working again, thx 👍

ZzyzxTek, 2023-05-12:
Reported same issue on TALONF7V2 in discord #supprt-forum.
Flashed PR12796 and no more lockup, motors spin fine.

I did not see this problem on my IFLIGHT_BLITZ_F722. I'll double check, maybe I flashed something other than zulu to it...

PR #12796 (2023-05-12): Fix bidirectional dshot on F7
Fixes: #12794
closed 2023-05-12 by haslinghuis