betaflight/betaflight #12797: Holybro Kakute F7 Mini V3 - BF4.4.X Random freezes/crashes/reboots to dfu while saving, can't write to or erase blackbox.

djc0ntra, 2023-05-11:
### Describe the bug

When using Betaflight firmware 4.4.0, 4.4.1, or 4.5.0 on a Holybro Kakute F7 Mini V3, Betaflight configurator's communication with the flight controller will randomly hang while trying to save changes to features on the Motors, Failsafe, and Blackbox menus. This also happens when trying to erase the blackbox flash using configurator, but seems to be successful in the CLI. To add to this, no blackbox files seem to record to the flash chip during flight even though logging is enabled. 

In order to continue once the communication is hung, I have to disconnect the USB cable and reconnect it in order to reconnect to the flight controller. Sometimes while doing this, the flight controller boots into DFU mode, resets its configuration, or even throw an error in windows stating "Unknown USB device (device descriptor request failed". When these last two issues happen, there is no way to reconnect to betaflight and the flight controller needs to be reflashed in DFU mode using the onboard boot button. 

None of these issues appear to be present in Betaflight 4.3.X. If I downgrade to any 4.3.X firmware, everything works as expected, including blackbox logging. 

### To Reproduce

Flash any betaflight 4.4.X or 4.5.X firmware on a Holybro Kakute F7 Mini flight controller.

### Expected behavior

Users of this flight controller should be able to save configurations in any menu in Betaflight Configurator without communication hanging, or the flight controller bricking itself. 

### Support ID

```txt
35395589-2063-4aeb-a2a6-ee2c6799db2b
```

### Flight controller

Holybro Kakute F7 Mini V3

### Other components

-DJI Air unit
-FrSky R-XSR
-Matek Systems M8Q-5883 GPS
-Holybro Tekko32 45A ESC
-iFlight XING2 1404 4600kv motors
-UPL 470uf 35V solid state capacitor

### How are the different components wired up (including port information)

# serial
serial 0 131073 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200

### Add any other context about the problem that you think might be relevant here

If you need me to run any tests or capture any logs just let me know.

haslinghuis, 2023-05-14:
Please retry 4.5.0 and 4.4.2 as we merged some changes

djc0ntra, 2023-05-18:
> Please retry 4.5.0 and 4.4.2 as we merged some changes

Hi Mark, 
Here are the results of retrying both versions on my flight controller:

BF 4.4.2:

-Custom defaults get applied after flashing
-Barometer and Blackbox unavailable
-When applying my drone's backed up configuration from 4.4.1 I see the following errors in the CLI:

# resetting to defaults
###ERROR IN resource: INVALID RESOURCE NAME: 'CAMERA_CONTROL'###
###ERROR IN set: INVALID NAME: baro_bustype = I2C###
###ERROR IN set: INVALID NAME: baro_i2c_device = 1###
###ERROR IN set: INVALID NAME: baro_hardware = BMP280###
###ERROR IN set: INVALID NAME: blackbox_device = SPIFLASH###

-Aside from that, the firmware works for a little while, but then begins to have the same intermittent freezing issues while saving and the flight controller needs to be powercycled to recover

BF 4.5.0 (using configurator 10.10.0)

-No custom defaults window pops up after flashing
-applying drone's configuration dump from 4.4.1 seems to work; barometer, gps, and mag are detected
-If you click save and reboot from the Configuration menu however, baro and mag hardware types get reset back to AUTO in the config, and no longer get
 detected. (This is probably a separate issue in the firmware)
-I get the same behaviour if I try the above in Configurator 10.9.0

I didn't try flying either because both seemed a bit iffy or unstable. BF 4.4.1 with Configurator 10.9 seems to be the most stable of the three, with everything working 95% of the time, but I still experience the flight controller rebooting into DFU mode from time to time after saving, and the blackbox doesn't consistently record, nor sometimes show how much space is used in configurator despite log files being present, and if there are log files present, they aren't readable in PIDToolbox, only Blackbox explorer meaning they could be corrupt or missing information when being recorded.

haslinghuis, 2023-05-18:
Reference: https://github.com/betaflight/betaflight/pull/12814#issuecomment-1552239476

After that PR is merged baro and blackbox should work again.
For now adding `BARO FLASH` to custom defines when flashing 4.4.2 should make them available..
Also add `Cam[era] Control` in Other Options.

djc0ntra, 2023-05-18:
> Reference: [#12814 (comment)](https://github.com/betaflight/betaflight/pull/12814#issuecomment-1552239476)
> 
> After that PR is merged baro and blackbox should work again. For now adding `BARO FLASH` to custom defines when flashing 4.4.2 should make them available.. Also add `Cam[era] Control` in Other Options.

Hi Mark,
I just tried this now, and everything seems to be working a lot better than before with those custom defines on 4.4.2. So far I've yet to experience any hanging while saving, or reboots into dfu mode, and blackbox logs seem to be created correctly. You can ignore my previous comment about PIDToolbox, as that seems to have been my issue with how I installed it, and seems to be fine after a reinstall.

If it means anything, when I run the defaults command in the CLI on 4.4.2 I still get the following errors, but they don't appear to affect anything that I am aware of:

# resetting to defaults
###ERROR IN resource: INVALID RESOURCE NAME: 'OSD_CS'###
###ERROR IN set: INVALID NAME: max7456_spi_bus = 3###

haslinghuis, 2023-05-18:
> ###ERROR IN resource: INVALID RESOURCE NAME: 'OSD_CS'###
###ERROR IN set: INVALID NAME: max7456_spi_bus = 3###

Thanks for testing. IS OSD working?

Please provide support ID

djc0ntra, 2023-05-18:
> > ###ERROR IN resource: INVALID RESOURCE NAME: 'OSD_CS'###
> > ###ERROR IN set: INVALID NAME: max7456_spi_bus = 3###
> 
> Thanks for testing. IS OSD working?
> 
> Please provide support ID

OSD is working, although sometimes when accessing the betaflight menu on my dji goggles v2, the osd won't clear the post-flight statistics before trying to draw the menu itself, causing the menu to draw over-top of the statistics table. If you want me to log a separate bug for that just let me know. Everything else seems to work fine after doing a few rounds of pid tuning flights today.

My latest support ID is: b35c1e41-fff0-4515-9c5a-5c1fad17dad1

haslinghuis, 2023-05-18:
Baro does not show up in `status` command

haslinghuis, 2023-05-18:
You can do a core only build to check the driver status

Check

    get altitude_source
    get altitude_prefer_baro

c3r63ru5, 2023-05-22:
I'm experiencing similar issues with my Holibro Kakute F7 (also F745), do you also see random hangs after/during bootup without USB connection? After connecting the battery, I can see the FC did not boot properly when I don't get OSD and cannot arm. Power cycling sometimes helps. I also see random hangs when saving PID's in the configurator.

djc0ntra, 2023-05-23:
> You can do a core only build to check the driver status
> 
> Check
> 
> ```
> get altitude_source
> get altitude_prefer_baro
> ```

It looks like I can do this on the build I have installed. These are the settings I'm currently using:

# get altitude_source
altitude_source = DEFAULT
Allowed values: DEFAULT, BARO_ONLY, GPS_ONLY

# get altitude_prefer_baro
altitude_prefer_baro = 100
Allowed range: 0 - 100

To see if I could narrow down the problem, I tried testing the following changes at my desk but couldn't reproduce the flicker on the altitude osd element:

set altitude_source = GPS_ONLY
set altitude_prefer_baro = 0

Then a second time with: 

set altitude_source = BARO_ONLY
set altitude_prefer_baro = 100

I'll probably have to try again while flying to see if there's any difference. 

In case it's helpful, the latest support id showing the barometer being detected correctly while on LIPO power is: 
8b7b9364-a57a-4b7c-b1e9-2b1e78034712

c3r63ru5, 2023-05-27:
In my case when OSD does not appear (also using MSP with WTFOS) the quad also would not arm, so pretty sure the FC hangs. So maybe some other issue and this here is only with USB being connected?

haslinghuis, 2023-05-30:
`4.4.1` won't work with this board. You need 4.4.2 (pre-release).

Choosing `OSD_HD` and leaving out `OSD_SD` and `VTX` should get rid of the MAX7456 and OSD Chip Select pin error.

DJI issues should be reported separate.

closed 2023-05-30 by haslinghuis