betaflight/betaflight #12798: HAKRC F4126 AIO - random gyro blips when rpm filtering enabled

joemanfpv, 2023-05-12:
### Describe the bug

HAKRC F4126 BLHELI_S 20A WHOOP AIO FC

Installed this AIO on an existing quad and experience throttle blips while flying.  It sounds like little chirps or tufts of air every few seconds when it happens frequently.  When holding the quad (carefully) while idling, there are clear throttle blips occurring every few seconds.

Blackbox logs show that there is a blip response on one axis (roll) that causes the motors to ramp up in an attempt to compensate.

These blips seem to not occur when RPM filtering is disabled.

Attached log was one of these 'hand tests' - it has 3 gyro blips starting at 10 seconds in.  This log also has DHOT motor rpm debug enabled.

[btfl_001_video.zip](https://github.com/betaflight/betaflight/files/11458878/btfl_001_video.zip)

I assume this is a software issue since the issue does not occur when RPM filtering is disabled.  

### To Reproduce

Flash Bluejay to ESC.  Configure 4.4.1 with RPM filtering.  

### Expected behavior

Expected to be nice and smooth...

### Support ID

```txt
?
```

### Flight controller

HAKRC F4126 BLHELI_S 20A WHOOP AIO FC, purchased from pyrodrone

### Other components

Caddx Vista, GPS on soft serial, crossfire nano RX.

### How are the different components wired up (including port information)

RX on port 1, vista on 2, GPS on soft serial

closed 2023-06-19 by github-actions[bot]

SynchroFPV, 2025-01-18:
Well, its happening to me now Exact same board exact same problem. Except my motor chirping went away as soon as I flashed all the esc to 0.21.1. But then a new issue arose, this time quad will suddenly bank 45 degrees right. Not sure if its my counteracting the roll with stick and about a sec after I get back normal control like nothing ever happened. Everytime when it banked 45 degrees it stops at the end of the bank then gitters for a split sec there then back to level. I tried to emulate it but failed. Its not at random either, cant explain. I'm not a beginner yet not an expert either, so i tried to explain as best as i could. Lastly, I looked at the specification and wiring diagram sheet that came with the FC and I noticed under ESC firmware is states BLHELI_S (G-H-20) WSC. Funny thing was ESC Configurator would not flash esc G-H-20. It always default back to G-H-30. Not sure if thats ok, but started up flew like normal, until I start doing acros, faster movements, more punch outs, power loops, etc and i know that sudden right 45degree bank is due soon. and ta-da. Right on que. I'm doing an entire fc swap now, cause i cant fly with a peace of mind with my quaD in this stage. but I have 2 of this HAKRC F4126 20A AIOs. The other one is still on default 4.5.1 betaflight settings as well but with BiDirectional Dshot still not turned on yet. Any advise will be golden, otherwise I'll just continue to search for answers until I cant. lol Thanks in advance to anyone helping and putting the effort to type and explain. Thank you.

Frame: Pavo20 Pro (2.2 inch props)
FPV: dji 03
motor: Miss 1203 6500KV
Cells: 4s
GPS: none
LED: none: (switched off tab in Configuration)

SynchroFPV, 2025-01-18:
> Well, its happening to me now Exact same board exact same problem. Except my motor chirping went away as soon as I flashed all the esc to 0.21.1. But then a new issue arose, this time quad will [...]

I forgot another thing i tried, i set DSHOT bitbang in CLI then switched off LED tab in Configuration....

TinoFpv, 2025-03-05:
@SynchroFPV I got the same issue, still on latest Bluejay firmware where he thinks G-H-30 is correct for it. (Im on G-H-20 now)
My little cineon c20 v2 on BF 4.5 is doing randomly blibs, at idle and when flying too. I tried EVERYTHING damn, can't get it what is wrong. You flashed the esc with the beta version G-H-20 0.21.0? Can't see a 0.21.1. Need a solution thanks.

TinoFpv, 2025-03-05:
SOLVED:
I needed to disable dshot bitbang via CLI:
set dshot_bitbang = OFF
now everything is ok happy flying!