betaflight/betaflight #12800: CRAZYBEEF4DX (STM32F411) Missing onboard blackbox in 4.4

phullivan, 2023-05-12:
### Describe the bug

Hi there,

I'm building a tinywhoop, and have purchased the [HappyModel Cross F4](https://www.happymodel.cn/index.php/2023/03/11/crossf4-elrs-1-2s-flight-controller-for-tinywhoop-with-bmi270-imu-and-serial-uart-elrs-v3-0-receiver/). This FC has onboard blackbox logging.

In a 4.3 firmware, blackbox logging is available, but after upgrading to 4.4 no blackbox device is available.

I am unable to get/set `blackbox_device` in the command line also.

### To Reproduce

- Flash CRAZYBEEF4DX to 4.4.1 or 4.4.0
- Navigate to the Blackbox Tab
- Blackbox logging is not available.

Note - flashing to 4.3.2 Blackbox logging returns.

### Expected behavior

Blackbox logging is available in the Blackbox tab in a 4.4 firmware.

### Support ID

```txt
# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 14 2023 / 14:37:16 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: HAMO, board_name: CRAZYBEEF4DX

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.33V, Core temp=62degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3752, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: f6bacc74f1c1d1db171dd6d9a229e807 (4.4.1)
System Uptime: 168 seconds, Current Time: 2023-05-12T08:50:32.435+00:00
CPU:10%, cycle time: 317, GYRO rate: 3154, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS ANGLE CALIB CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 14 2023 / 14:37:16 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4DX
manufacturer_id HAMO

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
[... 323 lines cut ...]
```
FLASH
FLASH_W25Q128FV
FLASH_M25P16
FLASH_W25N01G
FLASH_W25M
FLASH_W25M512
FLASH_W25M02G
```

None of these add the ability to access the blackbox

Flashing as a core only firmware, the blackbox logging ability returns.

Also, as a reference, the manual for the FC states that I must do this to enable to set the blackbox appropriately.

> Onboard blackbox enable method:
> 
> Type the following command to CLI:
> set blackbox_device = SPIFLASH
> set flash_spi_bus = 2
> resource FLASH_CS 1 a14
> save

I'm unable to do this because `blackbox_device` is an invalid name.

haslinghuis, 2023-05-12:
Flash a core only build and report back:

https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

phullivan, 2023-05-12:
Hey, thanks for following up. 

Flashing a core only build now detects the blackbox_device and the tab appears as expected.

Does this mean I should be using core only builds for this FC, or does that mean other hardware will not work?

haslinghuis, 2023-05-12:
No please now create a new support ID and post here to have the information I need to add the driver to the target configuration.

haslinghuis, 2023-05-12:
Output of status command in cli would be fine in this case.

phullivan, 2023-05-12:
Here you go, thanks so much for the help: 
```
# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 14 2023 / 14:37:16 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: HAMO, board_name: CRAZYBEEF4DX

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.33V, Core temp=60degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3752, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: f6bacc74f1c1d1db171dd6d9a229e807 (4.4.1)
System Uptime: 17 seconds, Current Time: 2023-05-12T10:55:09.549+00:00
CPU:10%, cycle time: 317, GYRO rate: 3154, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS ANGLE CALIB CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 14 2023 / 14:37:16 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4DX
manufacturer_id HAMO

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SERIAL_TX 1 A09
[... 258 lines cut ...]
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "6855a4c8-8f49-4a09-81f5-aac12f94db1d"

# copy ID and provide to the betaflight team.
```

haslinghuis, 2023-05-12:
Type the following command inc CLI:

```
resource FLASH_CS 1 A14
set blackbox_device = SPIFLASH
set flash_spi_bus = 2
save
```

phullivan, 2023-05-12:
Done. Blackbox status appears correct.

[image]

haslinghuis, 2023-05-12:
Can you please provide status output again

phullivan, 2023-05-12:
Here you go:
```
# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 14 2023 / 14:37:16 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: HAMO, board_name: CRAZYBEEF4DX

# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.33V, Core temp=31degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3752, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270
OSD: MAX7456 (30 x 13)
BUILD KEY: f6bacc74f1c1d1db171dd6d9a229e807 (4.4.1)
System Uptime: 18 seconds, Current Time: 2023-05-12T12:25:41.444+00:00
CPU:10%, cycle time: 317, GYRO rate: 3154, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS ANGLE CALIB CLI MSP

# flash_info
Flash sectors=256, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=16777216 JEDEC ID=0x00ef4018
Partitions:
  0: FLASHFS   0 255
FlashFS size=16777216, usedSize=0

# dump hardware

# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 14 2023 / 14:37:16 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4DX
manufacturer_id HAMO

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
[... 260 lines cut ...]
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 1
set scheduler_relax_osd = 1
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "4d5593c4-34b6-4ed2-a29e-062e0dc52772"

# copy ID and provide to the betaflight team.
```

closed 2023-05-12 by haslinghuis