betaflight/betaflight #12801: Ghost not working with latest master

TheIsotopes, 2023-05-12:
### Describe the bug

on latest master https://github.com/betaflight/betaflight/commit/c41b458e5891205da50465caeec0c1aa83beeb0c my MATEKF405MINITE with Ghost rx not working.

### To Reproduce

using Ghost and latest master on MATEKF405MINITE

### Expected behavior

working Ghost RX

### Support ID

```txt
Id: "c3b21df9-f056-486c-b1cf-6cf071b4f9f7"
```

### Flight controller

MATEKF405MINITE

haslinghuis, 2023-05-12:
@TheIsotopes I don't have Ghost. Was previous build okay (besides the bidir problem)?

TheIsotopes, 2023-05-12:
I'm still looking, but I think it's related to #12760.

haslinghuis, 2023-05-12:
@SteveCEvans please have a look.

closed 2023-05-14 by haslinghuis