betaflight/betaflight #12806: The drone flew by itself.

FlyFrg, 2023-05-13:
### Describe the bug

The second time my drone flies by itself. Before that, I checked the rescue mode settings. 10 times the drone returned to the take-off point without problems before this incident.

Watch this video to understand what happened:
https://youtu.be/EFNCMeBZCk4

### To Reproduce

Enable Angle Mode
Lose control signal
Watch as the drone flew about its business

### Expected behavior

Normal behavior

### Support ID

```txt
# version
# Betaflight / STM32H743 (SH74) 4.4.0 Mar  2 2023 / 07:30:29 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: HBRO, board_name: KAKUTEH7V2

# status
MCU H743 (Rev.V) Clock=480MHz, Vref=3.30V, Core temp=39degC
Stack size: 2048, Stack address: 0x2001fff8
Configuration: CONFIGURED, size: 3445, max available: 131072
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=BMP280
OSD: MSP (30 x 13)
BUILD KEY: 5c541f2178dacea4861b1968f2dc5fb4 (4.4.0)
System Uptime: 33 seconds, Current Time: 2023-05-13T12:34:05.528+00:00
CPU:8%, cycle time: 314, GYRO rate: 3184, RX rate: 15, System rate: 10
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00efaa21 128M
Arming disable flags: RXLOSS THROTTLE NOPREARM CLI MSP RPMFILTER

# flash_info
Flash sectors=1024, sectorSize=131072, pagesPerSector=64, pageSize=2048, totalSize=134217728 JEDEC ID=0x00efaa21
Partitions:
  0: BBMGMT    1003 1023
  1: FLASHFS   0 1002
FlashFS size=131465216, usedSize=131465216

# dump hardware

# version
# Betaflight / STM32H743 (SH74) 4.4.0 Mar  2 2023 / 07:30:29 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name KAKUTEH7V2
manufacturer_id HBRO

# resources
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B03
resource MOTOR 4 B10
resource MOTOR 5 A00
resource MOTOR 6 A02
resource MOTOR 7 C08
resource MOTOR 8 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 D08
resource SERIAL_TX 4 D01
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 D09
resource SERIAL_RX 4 D00
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C02
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_SCK 4 E02
resource SPI_SCK 5 NONE
resource SPI_SCK 6 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MISO 4 E05
resource SPI_MISO 5 NONE
resource SPI_MISO 6 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource SPI_MOSI 4 E06
resource SPI_MOSI 5 NONE
resource SPI_MOSI 6 NONE
resource ESCSERIAL 1 NONE
resource ADC_BATT 1 C00
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource PINIO 1 E13
resource PINIO 2 B11
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 A04
resource OSD_CS 1 B12
resource GYRO_EXTI 1 E01
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 E04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 A08
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer D12 AF2
# pin D12: TIM4 CH1 (AF2)
timer E09 AF1
# pin E09: TIM1 CH1 (AF1)

# dma
dma SPI_MOSI 1 13
# SPI_MOSI 1: DMA2 Stream 5 Request 38
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MOSI 5 NONE
dma SPI_MOSI 6 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_MISO 5 NONE
dma SPI_MISO 6 NONE
dma SPI_TX 1 13
# SPI_TX 1: DMA2 Stream 5 Request 38
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_TX 5 NONE
dma SPI_TX 6 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma SPI_RX 5 NONE
dma SPI_RX 6 NONE
dma ADC 1 8
# ADC 1: DMA2 Stream 0 Request 9
dma ADC 2 NONE
dma ADC 3 9
# ADC 3: DMA2 Stream 1 Request 115
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma TIMUP 1 0
# TIMUP 1: DMA1 Stream 0 Request 15
dma TIMUP 2 0
# TIMUP 2: DMA1 Stream 0 Request 22
dma TIMUP 3 2
# TIMUP 3: DMA1 Stream 2 Request 27
dma TIMUP 4 0
# TIMUP 4: DMA1 Stream 0 Request 32
dma TIMUP 5 0
# TIMUP 5: DMA1 Stream 0 Request 59
dma TIMUP 6 NONE
dma TIMUP 7 NONE
dma TIMUP 8 1
# TIMUP 8: DMA1 Stream 1 Request 51
dma TIMUP 12 NONE
dma TIMUP 13 NONE
dma TIMUP 14 NONE
dma TIMUP 15 NONE
dma TIMUP 16 NONE
dma TIMUP 17 NONE
dma TIMUP 0 NONE
dma TIMUP 0 NONE
dma TIMUP 0 NONE
dma pin B00 0
# pin B00: DMA1 Stream 0 Request 25
dma pin B01 1
# pin B01: DMA1 Stream 1 Request 26
dma pin B03 2
# pin B03: DMA1 Stream 2 Request 19
dma pin B10 3
# pin B10: DMA1 Stream 3 Request 20
dma pin A00 4
# pin A00: DMA1 Stream 4 Request 55
dma pin A02 5
# pin A02: DMA1 Stream 5 Request 57
dma pin C08 6
# pin C08: DMA1 Stream 6 Request 49
dma pin C09 7
# pin C09: DMA1 Stream 7 Request 50
dma pin D12 14
# pin D12: DMA2 Stream 6 Request 29
dma pin E09 12
# pin E09: DMA2 Stream 4 Request 11

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature GPS
feature OSD
feature AIRMODE
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 131073 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set baro_bustype = I2C
set baro_spi_device = 0
set baro_i2c_device = 1
set baro_i2c_address = 0
set adc_device = 1
set blackbox_device = SPIFLASH
set dshot_burst = OFF
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 109
set ibata_scale = 168
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 129,129,1,1
set pinio_box = 0,40,255,255
set usb_msc_pin_pullup = ON
set flash_spi_bus = 1
set gyro_1_bustype = SPI
set gyro_1_spibus = 4
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW0
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 0
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set i2c4_pullup = OFF
set i2c4_clockspeed_khz = 800
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "a9dd3c37-ecb6-43ad-a89a-ce3bd42a3a8f"

# copy ID and provide to the betaflight team.
```

### Flight controller

Kakute H7 V2

### Other components

GPS Matek m8q-5883

thetooth, 2023-05-13:
Were you able to recover the drone? Looks like GPS is giving false course over ground heading, the compass is continuously yawing to the right, but at the speeds you were going right before video loss I would expect the GPS to use positional error to derive heading. If you can, check the orientation of the GPS module, there should be an arrow on the PCB which needs to point towards the front. It's possible if the module is rotated 180 it would also fly off like this, but if it returned home safely before that's pretty unlikely.

FlyFrg, 2023-05-13:
> Were you able to recover the drone? Looks like GPS is giving false course over ground heading, the compass is continuously yawing to the right, but at the speeds you were going right before video [...]

The coordinates are showing correctly. The sensor is installed correctly. Before this case, gps rescue worked properly for 3 months.

thetooth, 2023-05-13:
Looking at source and trying to see if there is any way home point could be getting moved in flight, it seems unlikely. Fly away condition would have been set 15 seconds after high speed getaway so it would have gone a fair distance. I think it's a good idea to add a sanity check to see if distance to home delta has a high negative value for a short duration and to immediately attempt landing when this happens.

@FlyFrg I think you have a faulty GPS, the course over ground heading is wrong so betaflight will just blindly follow it. Best idea is to get another GPS and make sure home direction arrow is reliable. I think more pressing is to address the rx loss issue, you had good signal until suddenly it was gone, it then recovered but you did not regain control? perhaps elrs is a better option where you are located?

FlyFrg, 2023-05-13:
I don't know much about how GPS works, but isn't the direction of the take-off point calculated from the coordinates?
Control was not restored, although the signal was excellent. The drone did not respond to pressing the buttons on the remote control. I have already decided to change the receiver to Crossfire.

FlyFrg, 2023-05-13:
@thetooth I don't know much about how GPS works, but isn't the direction of the take-off point calculated from the coordinates?
Control was not restored, although the signal was excellent. The drone did not respond to pressing the buttons on the remote control. I have already decided to change the receiver to Crossfire.

thetooth, 2023-05-13:
The direction is a combination of the magnetometer and vector between last and current position, betaflight just reads that direction from the GPS since it's usually very stable. Once that rx loss message was gone any movement on the sticks should have cancelled failsafe so maybe it did not fully reconnect, I have heard the dji rc link can take a long time to reconnect but no experience with that myself.

FlyFrg, 2023-05-13:
I think that betaflight did not correctly calculate the vector, so the direction was not chosen correctly. And after all, there was a lot of time to correct the course and turn around, but the drone flew blindly where it needed to.

dz0ny, 2023-05-21:
I similar [issue](https://streamable.com/ty68aa). The video is lost, and then I trigger the rescue, I cannot get out of rescue (rc link is fine) by pressing any button (skipping disarm). I can still influence the direction a bit. After really doing a full left-right deflection it starts to respond and also detects that rescue mode is off. After this incident, I tested the rescue mode and it works fine as expected, I'll test the same scenario and flight path when the weather permits.

- Link is ELRS(50hz, LQ stays 100)
- Video is Walksnail
- Latest BF

haslinghuis, 2023-05-23:
> After really doing a full left-right deflection with sticks it starts to respond and also detects that rescue mode is off.

By design - after 30% stick deflection controller regains control if fail-safe state is cleared (and disables the rescue).

dz0ny, 2023-05-23:
> > After really doing a full left-right deflection with sticks it starts to respond and also detects that rescue mode is off.
> 
> By design - after 30% stick deflection controller regains control if fail-safe state is cleared (and disables the rescue).

However even after that rescue was still on.

closed 2023-07-01 by github-actions[bot]