betaflight/betaflight #12807: SPEEDYBEEF405V3 problem Blackbox

Mick51, 2023-05-13:
### Describe the bug

I updated to 4.4.1 and black box this but in error. On this FC it is an SD card reader.
In 4.5 Dev it works, and if I downgrade to 4.4.0 it also works. If I flash the core alone, it works too.

### To Reproduce

Try firmware upgrade

### Expected behavior

No blackbox will be available, including BLACKBOX in defines only enable configurator tab but on board memory will not be recognized.

### Support ID

```txt
ID : "51692859-9a33-4e8f-9252-85aa85184ff7"
```

### Flight controller

Speedebee F405 V3

Mick51, 2023-05-13:
version
# Betaflight / STM32F405 (S405) 4.4.1 Apr 10 2023 / 21:32:29 (e43d591b2) MSP API: 1.45

# config: YES
# board: manufacturer_id: SPBE, board_name: SPEEDYBEEF405V3

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=51degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3641, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=BMI270, ACC=BMI270, BARO=DPS310
OSD: MAX7456 (30 x 13)
BUILD KEY: b9cd90a43a78b121e960fac45c6d7124 (4.4.1)
System Uptime: 543 seconds, Current Time: 2023-05-13T14:18:43.302+00:00
CPU:15%, cycle time: 313, GYRO rate: 3194, RX rate: 15, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
# dump hardware

# version
# Betaflight / STM32F405 (S405) 4.4.1 Apr 10 2023 / 21:32:29 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF405V3
manufacturer_id SPBE

# resources
resource BEEPER 1 C05
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B08
resource MOTOR 4 B09
resource MOTOR 5 B00
resource MOTOR 6 B01
resource MOTOR 7 B05
resource MOTOR 8 B04
resource LED_STRIP 1 C09
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
[... 221 lines cut ...]
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

#

Mick51, 2023-05-13:
it looks like the resource is missing =>"resource SDCARD_CS 1 A15"
but when I try to add it manually it only tells me invalid resource...
error => " # resource SDCARD_CS 1 A15
###ERROR IN resource: INVALID RESOURCE NAME: 'SDCARD_CS'### "

haslinghuis, 2023-05-13:
Try: 

[image]

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis