betaflight/betaflight #12810: geprc gep-f722bt no gyro after upgrade to 4.4

fvtai, 2023-05-14:
### Describe the bug

my gepf722bt no gyro after upgrading to v4.4, reflash to older version also no luck..
pls advice how to solve this problem..
thanks

### To Reproduce

how and what to do with custom define

### Expected behavior

i reflash back older version and same no gyro

### Support ID

```txt
Id: "174afc7f-c94b-498a-a7b7-c07058cd9d60"
```

### Flight controller

geprc span gep-f722-bt

### Add any other context about the problem that you think might be relevant here

# copy ID and provide to the betaflight team.# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=58degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 3644, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked

OSD: MSP (30 x 13)
BUILD KEY: 50d0be9b602f59c4229c70e4cd4fb44b (4.4.1)
System Uptime: 1694 seconds, Current Time: 2023-05-14T04:28:39.145+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 11 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 1
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

haslinghuis, 2023-05-14:
Try core only build

https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

fvtai, 2023-05-15:
core only flashed, still no luck..

haslinghuis, 2023-05-15:
Can you provide a link to the product? There are 3 targets

Also try 4.4.2 instead:

[image]

haslinghuis, 2023-05-15:
According to the [manual](https://geprc.com/download/en/SPAN-F722-BT-Manual-V1_0.pdf) you would need 

Firmware target: EXF722DUAL

fvtai, 2023-05-15:
Ok i will try and let you know then..thanks

fvtai, 2023-05-15:
Hi, Firmware  EXF722DUAL flashed and it work...My bad wrong target firmware and thanks for your help.For your info, my flight controller only work with below than version 4.4.0 rc5 newer version won't work..usb cannot detect after apply custom default.Anyway once again thank you so much for your help.
TaiSent from Yahoo Mail on Android

Professor77, 2023-05-22:
Yes I tried 4.4.1 EXF722DUAL today and it was not detected after flashing. Will try 4.4.0 next.

haslinghuis, 2023-06-01:
Closing as fixed in 4.4.2 release.

closed 2023-06-01 by haslinghuis