betaflight/betaflight #12812: DYSF4PRO - no gyro found on every 2nd reboot (since BF4.3)

r0b985, 2023-05-15:
### Describe the bug

I have the DYS F4 PRO running Betaflight 4.1.1 and on every 2nd reboot it says "Warning: No Gyro/Acc found" and the reboot takes longer (~8s). 

This started with the release of Betaflight 4.3 (4.2 and below was fine). When I power the FC with a lipo, I have to quickly connect, disconnect and reconnect the battery in order to find the gyro - it's pretty annoying, expecially when setting things up.     

### To Reproduce

Either:
1) Connect the FC to betaflight
2) Click reboot on a tab that allows it

Or:
Plug in a battery over the main power connector

### Expected behavior

A quick reboot as usual and the gyro being found every time 

### Support ID

```txt
90379588-e7d9-4fa7-af39-82ece62840df
```

### Flight controller

DYS F4 PRO V1 from Banggood running the DYSF4PRO & OMNIBUSF4 target

Justicefpv, 2023-05-23:
One of these boards is currently sitting on my desk for testing. I'm also able to reproduce this problem.
This also happens on 4.2.2 Zulu, but not on 4.5 Zulu.

UlysseBE, 2023-07-18:
I have the same board and the same issue since BF4.3

lovejoy777, 2023-07-22:
> I have the same board and the same issue since BF4.3 with the OMNIBUSF4 firmware.
> I didn't try with the DYSF4PRO firmware, is there a difference ?

 it's happening on my dyst f4 pro FW too, since 4.3. as stated already it was fine on 4.2.11 and below.

haslinghuis, 2023-07-22:
@lovejoy777 Did you check all info in this room? According comment from @Justicefpv the issue should be fixed in 4.5.0

lovejoy777, 2023-07-22:
> @lovejoy777 Did you check all info in this room? According comment from @Justicefpv the issue should be fixed in 4.5.0 

oooo i missed that bit, thanks 👍

beyersito, 2023-08-22:
Hi! I am having the same issue with the old Dys Omnibus F4, can I update to 4.5 or it is not ready yet?

haslinghuis, 2023-08-22:
https://github.com/betaflight/betaflight/issues/12812#issuecomment-1559010488

klutvott123, 2023-09-02:
which gyro is it? if it's the MPU6000 I think it's fixed in master

klutvott123, 2023-09-02:
https://github.com/betaflight/betaflight/pull/12371

klutvott123, 2023-09-02:
Closing this. The info I can find on this FC points to MPU 6000, so it should be fixed in master(see post above). It would be great if anyone experiencing this issue could try master and report back.

closed 2023-09-02 by klutvott123

beyersito, 2023-09-03:
I can try it, any pointers on how to flash master/compile?

haslinghuis, 2023-09-03:
- Show release candidate
- Enable expert mode
- Select development
- Select 4.5-zulu

beyersito, 2023-09-08:
Tried it! Seems to be fixed 😄