betaflight/betaflight #12818: Unable to change rateprofile from LUA script.

wurtz, 2023-05-16:


klutvott123, 2023-05-16:
Did you change the pid profile at the same time as the rate profile before saving? I believe there's a limitation there that only lets you change either pid or rate profile at a time.

wurtz, 2023-05-16:
> Did you change the pid profile at the same time as the rate profile before saving? I believe there's a limitation there that only lets you change either pid or rate profile at a time.

No, Im a novice so I don't change PID at all. I just have a separate rate profile so thats the only setting I've been modifying (when moving from outdoor to indoor etc).

klutvott123, 2023-05-16:
> No, Im a novice so I don't change PID at all. I just have a separate rate profile so thats the only setting I've been modifying (when moving from outdoor to indoor etc).

The reason I'm asking is that in the last screenshot your pid profile is "2" while in the first one it's "1".  Please try changing only the rate profile while ensuring that the pid profile hasn't changed intentionally or unintentionally. I'll try to reproduce the problem too

wurtz, 2023-05-16:
> The reason I'm asking is that in the last screenshot your pid profile is "2" while in the first one it's "1". Please try changing only the rate profile while ensuring that the pid profile hasn't [...]

Thats a great catch. I just went back and watched the clip and I never actually change that PID setting. After re-entering the menu from save and exit, that PID profile is changed to 2 without me modifying it. I am currently traveling until the weekend so I'll have to repro this weekend when im back.

klutvott123, 2023-05-16:
It's easy to change pid profile unintentionally by holding the stick right just a little too long when entering that menu

wurtz, 2023-05-16:
> It's easy to change pid profile unintentionally by holding the stick right just a little too long when entering that menu

I did repro from the display lua script but fair enough. I'll test again when im back on Sunday. Thanks for your help here.

klutvott123, 2023-05-16:
I think the same logic applies when using both lua scripts, cms and msp

wurtz, 2023-05-16:
what does cms stand for?

klutvott123, 2023-05-16:
I have no clue lol. it's the menu in your goggles or "betaflight cms" lua script

haslinghuis, 2023-05-16:
Configuration Menu System

Was asking this myself: https://github.com/betaflight/betaflight/pull/1422

wurtz, 2023-05-21:
@klutvott123 thanks for keeping this open. Im back home and can confirm a couple test based on you're last ask. 

> Did you change the pid profile at the same time as the rate profile before saving? I believe there's a limitation there that only lets you change either pid or rate profile at a time.

1.) I've confirmed that I am not changing the PID and my rate profile continues to revert to rateprofile 3. The inconvenient and strange thing is that I was last set to rateprofile 1 via betaflight. So not only is this ignoring my change to rateprofile but it seems to be selecting its on profile and locking me to it. 

2.) I wanted to see if I could change any setting other than rateprofile. I modified my PID from 1 to 2 deliberately and exclusively from rateprofile. This setting changes and saves perfectly.

wurtz, 2023-05-22:
So I dug a bit deeper after seeing that this only affected rateprofile. I had rateprofile set to an adjustment switch unknowingly. This one can be chalked up to user error.  Thanks for your help, closing this out :)

closed 2023-05-22 by wurtz