betaflight/betaflight #12820: Handling on non-finite floats in code base

ledvinap, 2023-05-17:
### Describe the bug

When there is bug in code and invalid float operation is called, resulting NaN or Inf will propagate through code, eventually leading to unexpected results. Problem manifests quite far from invalid operation (or division by zero).

One example is #12815.
 
gcc `-ffast-math` enables `-ffinite-math-only'. Then compiler assumes that all results are finite, disabling some tests (`NaN == NaN` should return false, but in finite-math-only returns true / is optimized away). This makes problem a bit worse.

Disabling `-ffinite-math-only` is undesirable - correct BF code should not generate nonfinite results, so the optimization is valid and adds some preformace.

Possible measures to help with this:
1. Check FPSCR register (https://developer.arm.com/documentation/dui0646/a/Cortex-M7-Peripherals/Floating-Point-Unit/Floating-point-Status-Control-Register?lang=en)  periodically and indicate problem to user
 - OFC Overflow cumulative exception bit.
 - DZC Division by Zero cumulative exception bit.
 - IOC Invalid Operation cumulative exception bit.
2. Instrument code with isfinite() checks (+trap), enable this for debugging (-fno-finite-math)
 - gcc will optimize check in normal code
 - it will be a bit easier to find source of problem
3. Implement version of isfinite() that works in normal code (similar to #12815) and place checks in important places / mask known bugs until cause is found

Sadly, Cortex-M can't generate floating point exception/traps. This makes it harder to debug this problem.

### To Reproduce

#12815

### Expected behavior

To be decided

### Support ID

```txt
N/A
```

### Flight controller

All

ledvinap, 2023-05-17:
For iNav (https://github.com/iNavFlight/inav/pull/8953)
Enabling `-ffast-math` (or `-Ofast`) will optimize out all `isfinite()` check, for example in IMU code.

ledvinap, 2023-05-17:
Interesting reading: https://simonbyrne.github.io/notes/fastmath/

Prajwal-Koirala, 2023-12-27:
**I'm ready to assist with the bug analysis and potential solutions. Here's a breakdown of the key points:**

**Problem:**

- Invalid float operations (e.g., division by zero) generate NaN or Inf values.
- These values propagate through code, leading to unexpected behavior.
- Issue is compounded by `-ffinite-math-only` optimization, which masks NaN/Inf checks.

**Possible Solutions:**

1. **FPSCR Register Checks:**
   - Periodically monitor the FPSCR register for overflow, division by zero, and invalid operation flags.
   - Alert the user if any of these flags are raised, indicating a potential issue.

2. **Code Instrumentation with isfinite() Checks:**
   - Add `isfinite()` checks to relevant code sections for debugging.
   - Enable these checks only during debugging using `-fno-finite-math`.
   - Optimize away these checks in normal code for performance.

3. **Custom isfinite() Implementation:**
   - Create a custom `isfinite()` function that works in normal code (similar to PR #12815).
   - Strategically place these checks in critical code sections.
   - Use them to mask known bugs until the root cause is identified and fixed.

**Additional Considerations:**

- **Cortex-M Limitations:** Acknowledge the lack of floating point exception/traps in Cortex-M processors, which hinders debugging.
- **Expected Behavior:** Define clear expectations for how the code should handle invalid float operations to guide potential fixes.

**Next Steps:**

- Gather more details about the specific bug (e.g., reproduce steps, affected code sections).
- Assess the feasibility and impact of each solution approach.
- Prioritize solutions based on their effectiveness, performance implications, and ease of implementation.
- Implement chosen solutions with thorough testing to ensure correct behavior and prevention of unexpected outcomes.