betaflight/betaflight #12826: ARM command simulation

lackoder, 2023-05-20:
### Is your feature request related to a problem? Please describe

Hi!

I am developing a self-driving RC car that needs manual control too. I ordered a DJI Air Unit O3, because I have a DJI Remote Controller 2. Now, the radio will be connected to the AU and afaik the AU will transmit SBUS signals on its wires, that I am about to receive on a Raspberry Pi 4 (brain of the RC Car at the moment, will be upgraded for Jetson Nano later).
I heard that the new AU o3 will not switch to full power mode until it receives an ARM command.

Is there any chance that you can help me to develop a python script or some other way to send an ARM command to the AU? I know I could read thru the whole BF source to find it, but I am not into C language unfortunately and the code base is huge also.

Oh, the ARM command will be to enable full power video output for my DJI Googles 2 and vision. It would be not nice to always use it as in low power mode.

Cheers,
Laszlo

### Describe the solution you'd like

A python code or similar that sends ARM command and optionally DISARM on the wires.

### Describe alternatives you've considered

None.

haslinghuis, 2023-05-20:
Lots of python libraries available:

https://github.com/topics/multiwii
https://pypi.org/project/msp-python3/

closed 2023-05-20 by haslinghuis

jjziets, 2024-05-08:
@lackoder  did you ever get this to work? I had the same idea to use the Air unit 03, but for someone, I'm not getting Sbus commands from it. I assume one needs some FC connected to get it to give the DJI remote controller channels