betaflight/betaflight #12827: Dshot Signal May Stop on Disarm

ZzyzxTek, 2023-05-20:
### Describe the bug

I flashed the most recent BLHeli_32 32.9.9 test code to two different ESCs (Hobbywing_XRotor_BLHeli_32 and IFLIGHT_BLITZ_G2).
The 32.9.9 test code includes the modification for:

> Added protection against unintentional spoolups
> Input signal is qualified for 200ms before starting, and during the first 5s of running, signal timeout is only 40ms (vs generally 320ms).

My two quads were flashed with the most recent BF 4.5 zulu (as of 17 May).

On both quads, they boot up and fly perfectly fine
But after landing when I disarm the ESC gives the low/high throttle signal detected/zero throttle detected tones.
The same tones that are issued when the ESC first starts up and establishes comm with BF.

sskaug (on the BLHeli_32 RC Groups channel) replied:

> Thank you for reporting. I've tried to reproduce, but in my system I don't get the throttle detected/zero beeps upon disarming. But I think I understand why you get it in yours.
> 
> The updated 32.9.9 code (and 32.9.3 code) has the new protection against unintentional spoolups feature. Which means that for the first 5 seconds of spoolup (as well as after disarming), signal lost timeout is reduced from about 300ms to about 50ms. And most likely this is what happens in your case. BF for some reason stops sending signal (or sends invalid signal) for more than 50ms upon disarming. Which is not desirable. That being said, apart from the maybe annoying beeps, your system should work just fine.

I report this as an issue in case it points to something else going on that may need to be addressed. Other than the unexpected tones, this has not caused any operational issue with my quads.

### To Reproduce

1. Flash a BLHeli_32 ESC with either the 32.9.3 or 32.9.9 BLHeli_32 test code.
2. Flash the flight controller with Betaflight 4.5 zulu. Note I am uncertain what build matters for this.
3. Fly as normal.
4. On disarm, the startup tones will be issued. Note this may not happen on all ESCs.

See description above for more details.

### Expected behavior

Tones would not be issued on disarm.

However, this does not cause any operational issues that I've seen, just the unexpected tones. If this is the new normal, that's fine.

### Support ID

```txt
Id: "22a645de-752d-48d9-9317-0115a5e1c33f"
```

### Flight controller

IFLIGHT_BLITZ_F722 and TALONF7V2

### Other components

ESCs tested: Hobbywing_XRotor_BLHeli_32 and IFLIGHT_BLITZ_G2
BLHeli_32 testcode 32.9.3 or 32.9.9

### How are the different components wired up (including port information)

N/A

sanderpuh, 2023-06-16:
I have encountered the very same bug.
On top of that, sometimes I got RPMFILTER status flag, when this happened I had to reboot the FC and everything worked again.
 
The solution on my end was updating BLHeli_32 testcode 32.93 to testcode 32.95.
I have had no issues with the newer testcode so far.

ZzyzxTek, 2023-07-04:
on RC Groups, sskaug posted:
> We have now made a small mod in the 32.94 testcode that is now posted on github. When motor is stopped, timeout is set to 300ms, it is only set to 50ms for the first 5 sec of spinning.

With that newer 32.95 test code, I no longer get the additional tones after disarming with the two ESCs (Hobbywing_XRotor_BLHeli_32 and IFLIGHT_BLITZ_G2) where I was getting them with 32.99.

I also updated to a more recent Betaflight 4.5 nightly build as well, so I guess it's possible something was updated in BF...

@haslinghuis  Would you like me to close this issue, since I'm no longer experiencing it, or do you want to leave it open for now?

haslinghuis, 2023-07-04:
Interesting the BLHELI32 update is indeed fixing the issue [as expected], but some [new] issues were reported after some updates in BF firmware with mainstream esc firmware,

SupaflyFPV, 2023-07-04:
ok leave the bug active until I rebuild an F4 target...will try and get it done this week to follow up this one...

haslinghuis, 2023-08-09:
@SupaflyFPV any update on this?

SupaflyFPV, 2023-08-09:
> @SupaflyFPV any update on this?

ok I just flashed the latest master and hey presto - the bug is gone

happily this issue from me can be closed then...:

https://github.com/betaflight/betaflight/issues/12828

haslinghuis, 2023-08-09:
@ZzyzxTek please test on your side with 4.4.3-zulu / 4.5-zulu - so we can close this

ZzyzxTek, 2023-08-10:
> @ZzyzxTek please test on your side with 4.4.3-zulu / 4.5-zulu - so we can close this @SupaflyFPV closed #12828 as requested.

Tested and confirmed works with 4.4.3-zulu and 4.5-zulu.

I think the original issue I mentioned with the extra low/high throttle signal detected/zero throttle detected tones on disarm was fixed with the BLHeli_32 32.95 testcode update.

But nothing in BF has "unfixed" it!