betaflight/betaflight #12828: Higher Idle and Arming bug

SupaflyFPV, 2023-05-22:
### Describe the bug

Latest Master flashed today - on an Aikon F4 FC - I had an issue which has occurred before when work was done on the Dshot area. 

1. Arming can only be done once - then needs powercycle
2. Idle throttle is much higher than it should be on hovering

Flashing 4.4.1 or 4.4.2 works as normal. Latest Master the issue occurs.

Was using blheli32 32.7 - decided to flash 32.9 to check - no change. Only reverting back to 4.4 brings normal behaviour.

I can check soon with normal idle.

[JMA4.txt](https://github.com/betaflight/betaflight/files/11526241/JMA4.txt)

### To Reproduce

As per above

### Expected behavior

As above

### Support ID

```txt
diff attached
```

### Flight controller

Aikon F4

tbolin, 2023-06-13:
I found some similar cases on discord
> On the FLYWOOF405NANO after installing 4.5-Zulu I get an RPM Filter OSD warning preventing arming after I arm once, disarm and then try to re arm. This is also happening on a few other boards as well. Is this a thing that will get worked out before RC or Final or is it something needing looking into? I've had it happen with past dev builds and it's gone away in later releases. Just wanting to double check.

Another user also had the same problem on a TMOTORF4.

haslinghuis, 2023-06-16:
Please try with `dshot_bitbang = ON`

ASDosjani, 2023-06-23:
I have the same experience with NEUTRONRCF435MINI aio, latest AM32 firmware, dshot_bitbang = ON. After first disarm all motors has 100% dshot error.

haslinghuis, 2023-06-23:
@ASDosjani please check for a possible fix for AM32 [here](https://github.com/betaflight/betaflight/pull/12912)

haslinghuis, 2023-07-04:
@SupaflyFPV please check with BLHELI32 32.95 test code?

https://github.com/betaflight/betaflight/issues/12827#issuecomment-1620639122

haslinghuis, 2023-08-09:
Fixed in 4.4.3-zulu / 4.5-zulu

nerdCopter, 2023-09-06:
see also https://github.com/betaflight/betaflight/issues/13068