betaflight/betaflight #12829: Lumenier Lux Mini F7-missing blackbox feature?

GreenWeedle, 2023-05-23:
### Describe the bug

The FC has a 16M blackbox, but it doesn't appear in configurator?  Also, the gyro was set to 180, but needed to be changed to default to correct orientation in 3d model in confiurator.

https://www.getfpv.com/electronics/flight-controllers/mini-flight-controllers/lumenier-lux-mini-f7-flight-controller.html

### To Reproduce

install BF 4.4   (I believe it showed up previously in an older BF version)

### Expected behavior

no BB visible

### Support ID

```txt
*** Support data submitted *** 
Id: "162bc76d-ba90-46db-bab7-c32508564289"

# copy ID and provide to the betaflight team.
```

### Flight controller

LUXMINIF7

### Other components

BF4.4.1

haslinghuis, 2023-05-23:
Can you try:
 
[image]

GreenWeedle, 2023-05-23:
Thanks Mark!  This sems to have recovered the 16M Blackbox in configurator.  I'll make a test flight today to see if it's working.

The 1st Gyro is still defaulting to "CW180" but needs to be "Default" to correct orientation in the 3d model.

Cheers,
Brian

[image]<https://user-images.githubusercontent.com/8344830/240082524-87803859-4c7c-47b9-93b6-cf2e207478bd.png>

haslinghuis, 2023-05-23:
Roger, need to update in configuration
`set gyro_1_sensor_align = CW180  to CW0`

Does this need to stay as is?
`set gyro_1_align_yaw = 1800`

GreenWeedle, 2023-05-24:
Hi Mark,

Yes, it seems the gyro align yaw is indeed 1800 when I dumped my cli and this functions properly.

FYI, not sure if it it matters but I had to use the boot button to get into DFU and flash 4.4.2 Zulu….maybe this is to be expected, but it wouldn’t go there automatically when I tried to flash after downloading the cloud build or any other way I tried in software (cli “bl” command or the dfu button in configurator). Maybe this is normal, but wanted you to know. :)

Thanks much!
Brian

closed 2023-05-30 by haslinghuis