betaflight/betaflight #12832: Cloud build for MATEKF405TEMINI not working

TheIsotopes, 2023-05-23:
### Describe the bug

I can't build MATEKF405TEMINI with cloud build and current master.
On the screenshot you can see the highest build branch I can create in the cloud build.

[image]

This problem does not occur with any other Matek FC!?

### To Reproduce

try MATEKF405TEMINI cloud build with latest configurator / master

### Expected behavior

working cloud build for MATEKF405TEMINI

### Support ID

```txt
not possible
```

### Flight controller

MATEKF405TEMINI

haslinghuis, 2023-05-23:
@blckmn please have a look

haslinghuis, 2023-05-24:
Seems to be sorted:

[image]

closed 2023-05-24 by haslinghuis