betaflight/betaflight #12833: Baro: BMP280 is not recognised on SKYSTARSH7HD on 4.4.0, 4.4.1 and 4.5 dev build as well

YogeshAuto, 2023-05-23:
### Is your feature request related to a problem? Please describe

Baro: BMP280 is not recognised on SKYSTARSH7HD board.
It is a brand new H7 board with onboard BMP280.

*** Support data submitted *** 
Id: "a021c581-42b9-4928-912e-53d2c2d0c767"

### Describe the solution you'd like

need a fix to recognise barometer or provides steps/cli commands for settings if required.

### Describe alternatives you've considered

Tried 4.4.1 and 4.5 dev build as well with core only flashing but baro is not recognised.

### Other information

2023-05-23 @21:54:34 -- OS: Windows

2023-05-23 @21:54:34 -- Configurator: 10.9.0 (697943b)

2023-05-23 @21:54:38 -- Loaded release information for configurator from GitHub.

2023-05-23 @21:54:41 -- Serial port successfully opened with ID: 1

2023-05-23 @21:54:41 -- MultiWii API version: 1.45.0

2023-05-23 @21:54:41 -- Flight controller info, identifier: BTFL, version: 4.4.1

2023-05-23 @21:54:41 -- Running firmware released on: Apr 11 2023 00:24:24

2023-05-23 @21:54:41 -- Board: SKST/SKYSTARSH7HD(STM32H743), version: 0

2023-05-23 @21:54:41 -- Unique device ID: 0x36002e3430511236333333

2023-05-23 @21:54:41 -- Craft name: YSDrone

2023-05-23 @21:54:41 -- Arming Disabled

2023-05-23 @22:01:16 -- CLI mode detected

2023-05-23 @22:06:27 -- Serial port successfully closed

2023-05-23 @22:06:34 -- Serial port successfully opened with ID: 2

2023-05-23 @22:06:34 -- MultiWii API version: 1.45.0

2023-05-23 @22:06:34 -- Flight controller info, identifier: BTFL, version: 4.4.1

2023-05-23 @22:06:34 -- Running firmware released on: Apr 11 2023 00:24:24

2023-05-23 @22:06:34 -- Board: SKST/SKYSTARSH7HD(STM32H743), version: 0

2023-05-23 @22:06:34 -- Unique device ID: 0x36002e3430511236333333

2023-05-23 @22:06:34 -- Craft name: YSDrone

2023-05-23 @22:06:34 -- Arming Disabled

2023-05-23 @22:09:00 -- CLI mode detected

2023-05-23 @22:09:09 -- Chrome API Error: User cancelled.

2023-05-23 @22:10:49 -- Serial port successfully closed

2023-05-23 @22:11:29 -- Serial port successfully opened with ID: 3

2023-05-23 @22:11:39 -- No configuration received within 10 seconds, communication failed

[... 707 lines cut ...]
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set i2c4_pullup = OFF
set i2c4_clockspeed_khz = 800
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "a021c581-42b9-4928-912e-53d2c2d0c767"

# copy ID and provide to the betaflight team.

haslinghuis, 2023-05-23:
Also check 4.4.2 pre-release

Baro is using I2C2  in unified target and SPI2 in config.h so we need to determine what bus type is being used.

https://github.com/betaflight/unified-targets/blob/master/configs/default/SKST-SKYSTARSH7HD.config
https://github.com/betaflight/config/blob/master/configs/SKYSTARSH7HD/config.h

closed 2023-05-23 by haslinghuis

YogeshAuto, 2023-05-24:
Thanks, baro got recognised in 4.4.2