betaflight/betaflight #12836: UART6 instead of USB on F411 boards

annoyinganime, 2023-05-25:
### Is your feature request related to a problem? Please describe

Looking through the datasheet of STM32F411 mcu I've found one interesting thing: it was always told that f411 has two UARTs, and additional serial devices need to be connected using SoftSerial (if you wanted to connect ELRS + SmartVTX + GPS you would often connect ELRS and GPS on hardware uarts for reliability, and vtx on a softserial pin). However in the datasheet it is shown that there are actually three UARTS - 1, 2 and 6, with UART 6 pins shared with USB, so with the default firmware it is not used.

### Describe the solution you'd like

With the rise of popularity of digital FPV systems software UART is no longer a nice option to use, because baud rates need to be higher, and it puts way more load on a not that powerful F411 MCU. What I am proposing is F411-UART6 builds, that need to be configured using USB-UART adapters, with MSP enabled by default on UART6. This will make both setup and usage not that difficult - for configuration you will need special USB adapter, and after you've configured everything you need you'll just unplug usb, plug there your VTX TX and RX and go out to fly - no reconfiguration for VTX needed, because it also uses MSP protocol. (so basically returning to the good old days of SP Racing F3, where you'd place MicroMinimOSD on UART 1 shared with CP2102 usb-uart converter)))
With new addition of remote build it can be added as build option, so I think it would not be that difficult to implement

### Describe alternatives you've considered

Softserial for GPS is not a good option either - with recent rise of M10 chips they also need fast UART for best performance.
An alternative would be to use different board with more capable f405 or higher MCU, however there are some boards that don't have any competition in both size and weight, that only are made with f411 chips (ghf13aio - 16*16 board with 13a 4s ESCs; Happymodel CROSSF4 ELRS for 1-2s long range builds with GPS RTH)

annoyinganime, 2023-05-25:
Just searched through betaflight code - seems like betaflight already supports UART6 on f411 targets, will try to find what i need to change to compile it

annoyinganime, 2023-05-25:
YES IT WORKS.

I needed to edit target config file to disable USB detect pin and enable UART6 tx and rx, and also unified target .c file to disable VCP.
So now the only thing left is to add build option in cloud build 

[image]
[image]

haslinghuis, 2023-05-25:
MSP needs to be locked - so user can't disable configuration

haslinghuis, 2023-05-25:
@jflyper

annoyinganime, 2023-05-26:
> MSP needs to be locked - so user can't disable configuration

You can't actually disable configuration completely - betaflight sets MSP to first available UART if it detects that there is no MSP configured (just tested it by disabling all MSP). 
But yes, it is inconvenient to solder wires to uart 1 every time you mess up the configuration

annoyinganime, 2023-05-28:
After reading more of the source code and thinking about it i have more interesting idea: 
we already can detect USB connection from PC using USB_DETECT pin; what if we will enable USB_VCP when we detect incoming voltage on that pin, and enable UART6 all the other times? This will make set up way easier, because you'll no longer need any adapter!

closed 2023-07-06 by github-actions[bot]