betaflight/betaflight #12842: Omnibus F4 Target - Blackbox functionality missing on 4.4.1

Ddraig62SPD, 2023-05-28:
### Describe the bug

Hi. Flashed BF4.4.1 via 10.9.0 configurator to a HGLRC Flame F4 FC (Omnibus F4 target) which was previously running Blackbox normally on the 16MB internal flash chip. All tabs seemed to be running OK in BF 4.4.1 but when accessing the Blackbox tab the page is blank with a message 'Your FC's firmware does not support blackbox logging'.

### To Reproduce

Flashed with LED, OSD(HD), PIN I/O, VTX build options. 
Full chip erase (ON), Radio Protocol = CRSF/GHST/SBUS, Motor Protocol = DSHOT ESC, Telem Procol = None. Custom Defines = None.
Defaults applied on FC reboot.
Running ELRS RC Link, O3 FPV.

### Expected behavior

All tabs seemed to be running OK in BF 4.4.1 except for the Blackbox tab. Expected to see options for logging options, sample rate, erase flash but the tab was blank with a message 'Your FC's firmware does not support blackbox logging'. 

### Support ID

```txt
e8e254ee-3754-4c0a-b0be-8c56ab2a407f
```

### Flight controller

HGLRC Flame F4 OSD PDB (Omnibus F4 target). Source = ??

### Other components

Frame: Transtec Frog Lite 5".
FC: HGLRC Flame F405, BF V4.4.1.
ESCs: Racerstar RS30AL 30A Bi-directional Dshot600 BlueJay V0.16
Motors: Racerstar 2306S/2400Kv
Receiver: ELRS Namimno Nano 2.4GHz V3.0.1
Transmitter: Jumper T-Lite V2 EdgeTx 2.71, ELRS internal module V3.0.1. 
FPV: DJI O3 backpack system & V2 Goggles (latest May'23).
Other: Matek TVS anti-spark, ESR capacitors 2x 470uF 35V, passive buzzer.
Props: DAL T5040x3 Cyclone.

### How are the different components wired up (including port information)

UART 1 - MSP Displayport to DJI O3 Air Unit
UART 3 - Serial RX to discrete ELRS RX
UART 6 - Not used.
Buzzer to dedicated FC BZ+/- pads.
Discrete ESCs x4 wired to dedicate S1-4, +/- pads.
TVS Diodes & 470uF ESR cap soldered to XT60 socket, Separate 470uF ESR at pigtail connection to FC power pads.

### Add any other context about the problem that you think might be relevant here

Blackbox functionality was visible & working in BF4.3.1. Have not tried to reflash as I was most of the way through my configuration when I noticed the issue. Could not see any Blackbox specific build options for the 4.4.1 firmware in configurator or custom defines.  I was unable to flash any 4.4.1 firmware with the 'no reboot sequence' option enabled (this normally works on my Macbook Pro 2012 running Catalina). FC simply hung with no progress bar display. Looked like the USB port could not be accessed to start flashing so simply disconnected. Only disabling the 'no reboot sequence' option allowed a successful flash. I've built 30+ multirotors since my first KK2 FC build so normally figure out the root cause ... but I noticed that others have seen similar Blackbox issues on 4.4.1 so figured I'd give the dev's some more evidence to help with a possible fix.
 
Combined CLI Status & Diff_All file attached FYI. 

Hit me up if you need further info. 
Thanks in advance & keep up the good work :)

[BTFL_cli_Frog_diff_all_4.3.1_20230528_111543.txt](https://github.com/betaflight/betaflight/files/11586044/BTFL_cli_Frog_diff_all_4.3.1_20230528_111543.txt)

haslinghuis, 2023-05-28:
https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

Ddraig62SPD, 2023-05-28:
Thanks Mark. I’ll try the Flash custom define tomorrow and will keep you
posted. Cheers :)

McGiverGim, 2023-05-29:
The HGLRC F4 Flame has it's own target, but I'm not too sure if you need the HGLRCF405 or HGLRCF405AS. @haslinghuis are you sure he needs to use the Omnibus one with custom defines?

Ddraig62SPD, 2023-05-29:
Hi Guys. I tried reflashing the OMNIBUS F4 target with FLASH in the
Custom Defines list. The BlackBox tab in Configurator 10.9 is now populated
and active; however the internal on-board flash does not appear as a
selectable option in the 'Blackbox Logging Device' drop-down (only 'No
Logging' or 'Serial Port' appear), so I'm unable to use the internal flash.

FYI - I attempted to flash with the HGLRCF405/HGLRCF405AS targets but the
configurator threw up a 'firmware mismatch' warning as the FC is identified
as an OMNIBUS F4, so I aborted the operation. As mentioned, the
blackbox functionality (and all other features) worked correctly when
running the FC on BF V4.3.1 with the auto-detected OMNIBUS F4 target.

I'll just make do without the blackbox functionality until a possible fix
is found, as I need V4.4.1 to run the HD OSD on the O3 system. I'll just
load one of the tune presets for 5" as there's plenty of choice :)

Cheers, Steve

haslinghuis, 2023-05-29:
Please try the development pre-release 4.4.2 without custom defines. Also try the core only build.
We need a support ID (you can get it on the CLI tab)

Ddraig62SPD, 2023-05-29:
Hi Mark. V4.4.2 (latest 4.4-maintenance version) seems to have resolved the
issue. On-board flash is now recognised & erase options are visible. I'll
revert to 4.4.1 in the interim until 4.4.2 is officially released. Support
ID below FYI.

Thx & best regards, Steve

*** Support data submitted ***
Id: "c05a7878-74a3-41c4-9071-75037169f65a"

haslinghuis, 2023-05-29:
Thanks, JEDEC ID confirmed.

Closing as resolved in 4.4.2 (pre-release)

closed 2023-05-29 by haslinghuis

Ddraig62SPD, 2023-05-29:
Many thanks :)