betaflight/betaflight #12843: DJI Integra & Vista Flywoo LR2 BF 4.4.1 OSD lag/stutter

AlexandruIspas89, 2023-05-29:
### Describe the bug

The HD OSD overlay with DJI Integra and Vista lags behind. Once the flight controller is connected to the BF configurator 10.9 and under ports UART 3, VTX (MSP + DisplayPort) is changed from auto to 250000 baud rate, the lag is gone. If the drone is left for more than 5 minutes without power the lag comes back up. On the same setup there was no lag in the past, issue started with BF 4.4.1 and 4.4.0. Lag also affects GPS rescue as it is not able to fly back correctly. If lag is not present everything works correctly. CPU load is around 52%. 

https://pastebin.com/A4yMXYNA 
https://youtu.be/Kx8jSVqC1YE 

### To Reproduce

Connect the battery to the flywoo explorer and move the drone, observe the roll and pitch in the goggles. They respond after a few good seconds. 

### Expected behavior

The baud rate setting under MSP Display port should be remembered even after some time without the flight controller powered. There could be an issue with this function. I tried also on a second flywoo explorer lr 2 and the same behaviour is there. 

### Support ID

```txt
Id: "a1a27781-8987-4dd3-9bdc-4d7df5530975"
```

### Flight controller

Goku F745 Nano stack 16x16 

### Other components

Flywoo explorer LR 2 with caddx vista, M8N mini gps v2.0 and Goku F745 nano stack 16x16, tbs crossfire nano receiver.

Dji integra and caddx vista are updated on the last available software that supports HD canvas. 

### How are the different components wired up (including port information)

UART 1 - crossfire
UART 2 - GPS 
UART 3 - Caddx Vista

haslinghuis, 2023-05-29:
Please check with 4.4.2 and 4.5.0 pre-release

AlexandruIspas89, 2023-05-29:
I tried with both 4.4.2 and 4.5.0 and the OSD still lags. It must be related to the new DJI HD canvas mode. One time I managed to get it working correctly with 4.4.1 following the procedure to first turn the drone one and after the radio and Dji Integra. After first flight on the second run it started to lag.

macgyverek, 2023-06-07:
I observed the same issue after update to (first fly) 4.2.1. I had stuttering on osd, but only on one flight.

dxs94, 2023-06-25:
Did you try lowering OSD framerate? That seems to have done it for me with a similar issue. "osd_framerate_hz" with a default of 12. I lowered to 6.

74Ch1m, 2023-07-28:
I had the same issue on an iFlight F745 AIO on 4.4.1 and 4.4.2. Everything was lagging like 20 secs behind. And at sometime the OSD started flashing heavily.
I had a lot of OSD elements including realtime and date. I removed this exact same element and lowered the osd refresh rate to 6 hz as mentioned above. It worked for me. Don't know if it was to many OSD elements or the lower refresh rate. Maybe both :-)

AlexandruIspas89, 2023-07-30:
Hi thanks for the advice, I did lower the OSD refresh rate from 12 to 6 Hz and now the lag is gone. I did not fly the drone yet as it undergoes some upgraded but I did a workbench test and it looks normal.

closed 2023-07-30 by AlexandruIspas89