betaflight/betaflight #12860: 4.4.2 Dshot motor direction commands not working after #12709

limonspb, 2023-06-03:
### Describe the bug

Motor direction wizard stopped working after #12709 (at least on F7X2 (Foxeer F7 V4))
IFLIGHT F4 seems to be working.
Also a couple of reports from people on facebook and in BF discord.

### To Reproduce

Flash 4.4.2
Activate DSHOT 600 or DSHOT 300
Click save and reboot
Try to use motor direction visard tools to change motor direction

### Expected behavior

Expected behavior: motor direction changes.
Actual behavior: motor direction does not change.

### Support ID

```txt
47c439df-556f-4461-b08a-dd595d3842c8
```

### Flight controller

FOXEERF722V4

### Other components

Did a quick git bisect on 4.4-maintenance branch, it showed that the bug happens after #12709:
@SteveCEvans could you please have a look?
[image]

### How are the different components wired up (including port information)

diff can be pretty much empty:
[image]

sugaarK, 2023-06-03:
Happening with bluejay and am32 as well

limonspb, 2023-06-03:
Bisected master branch, it's this PR:
https://github.com/betaflight/betaflight/pull/12685

LordGG, 2023-06-03:
Same issue here with Foxeer F722 v4 MPU6000 + Foxeer ESCs Reaper 65A.

Change motor direction not working anymore with 4.4.2 release. Works fine with 4.4.1.

I have also noticed that with 4.4.2 motors 3 and 4 get between 0 and 0.04% erros, while I have 0% error all the time with 4.4.1.

haslinghuis, 2023-06-03:
https://github.com/betaflight/betaflight/suites/13348912628/artifacts/729946040

LordGG, 2023-06-03:
> For testing:
> 
> 1. Save a dump
> 2. Flash https://github.com/betaflight/betaflight/suites/13348912628/artifacts/729946040
> 3. Restore dump

Setting motor direction is working again with this 4.4.3 firmware.

But I still get a tiny amount of errors on motors 3 and 4. It is below 0.1% but I have a rock solid 0% with 4.4.1, so there is still something going on I think.

nerdCopter, 2023-06-09:
is there a PR for bugfix yet?

limonspb, 2023-06-09:
> is there a PR for bugfix yet?

yes sir:
https://github.com/betaflight/betaflight/pull/12865 - master
https://github.com/betaflight/betaflight/pull/12866 - 4.4-maintenance

haslinghuis, 2023-06-10:
Think @nerdCopter was asking about the fix for the RPM errors we have since 4.4.2

Hopefully @SteveCEvans can find something causing this.

Errors need to be zero - rock solid - as before. It's a good measure to see anything is wrong as it's one of our unique selling points.

haslinghuis, 2023-06-13:
Fix has been updated and issue with error rates does not occur on two test subjects.
 
Please test development 4.5.0 with `#12865` in the commit field for testing telemetry errors.

limonspb, 2023-06-13:
> Fix has been updated and issue with error rates does not occur on two test subjects.
> 
> Please test development 4.5.0-zulu [pre-release] for testing telemetry errors.

is there a specific PR that fixes error problem? or it just somehow fixed itself?

haslinghuis, 2023-06-13:
@limonspb in this PR 

[image]

The PR solves the issue mentioned in the title. The telemetry error is another issue, see also #12828

LordGG, 2023-06-13:
> Please test development 4.5.0-zulu [pre-release] for testing telemetry errors.

I have just flashed 4.5.0-zulu [pre-release] (Latest), I still see a bit of errors on motors 3 and 4.

But I have a much bigger problem now : since I have flashed 4.5, my FC cannot enter Boot Loader mode anymore. Trying to flash the FC, using the boot button or typing "bl" in CLI ends up with a "Failed to open serial port" and a Windows Error message "unknown USB device". No more DFU mode, I cannot go back to 4.4 :'( (it can steel connect in normal mode).

The bootloader led is solid red, i have tried another identical FC, BF detected DFU Mode as usual.

Any idea how to get it back working ? It is just 2 weeks old ^^'

LordGG, 2023-06-14:
> O well the PR was solving another issue - as I couldn't reproduce on two test builds. So we will keep investigating.
I wasn't supposed to select a specific commit ? I just went with the latest, hope it was ok.

> Use boot button to get into DFU manually
> Restart (but recommend trying latest nightly configurator build) - sometimes just rebooting computer helps.
I did all that (boot button and reboot) without success. With no hope, I changed USB port and cable, could connect in DFU Mode and flash 4.4.1 again. I went back to first USB port and old cable, BL back working... It makes no sense to me, but it is working :)

LordGG, 2023-06-14:
> > O well the PR was solving another issue - as I couldn't reproduce on two test builds. So we will keep investigating.

I wasn't supposed to select a specific commit ? I just went with the latest, hope it was ok.
 
> > Use boot button to get into DFU manually
> > Restart (but recommend trying latest nightly configurator build) - sometimes just rebooting computer helps.

I did all that (boot button and reboot) without success. With no hope, I changed USB port and cable, could connect in DFU Mode and flash 4.4.1 again. I went back to first USB port and old cable, BL back working... It makes no sense to me, but it is working :)

haslinghuis, 2023-06-14:
To narrow it done here is the list of PR's which got into 4.4.2

Please notice the merge date - I would start with:
- https://github.com/betaflight/betaflight/pull/12671 (13-04)
- https://github.com/betaflight/betaflight/pull/12709 (22-04)

https://github.com/betaflight/betaflight/pulls?q=is%3Apr+milestone%3A4.4.2+is%3Aclosed

LordGG, 2023-06-14:
> To narrow down the issue with the bidirectional telemetry error - here is the list of PR's which got into 4.4.2 - so you might try flashing in order of merge:

Thanks ! I have noticed a different behaviour whether I just type the PR number or I actually select an entry in the list. I give you both results (with check out from the log) : 

#12671 --> OK 💯 
Checking out the branch -> CLOUD_HEAD (for pull/12671/head)
Fix SDC (#12671) --> OK 💯 
Checking out the branch -> CLOUD_HEAD (for aa228ea583b03a555a261c05e259493f55ac0f0d)

#12707 --> OK 💯 
Checking out the branch -> CLOUD_HEAD (for pull/12707/head)
Fix handling of attribute type (#12707) --> ERROR 👎 (same motor errors as before)
Checking out the branch -> CLOUD_HEAD (for 5fdcfb5bc5ab8391e81e9aed86ecc0c8c63f7945)

Since I had errors with #12709, I also tried the previous PR (#12707) which also had the errors.

I hope it can help. And if you can explain which one is wrong between typing just the PR number or selecting an actual entry in the list. I would have expected the same result.

haslinghuis, 2023-06-14:
Thanks for testing @LordGG 

Please share details about the hardware used on your build. Please also provide support ID (press button on CLI tab and provide the ID here).

Seems 12709 was merged before 12707 (so 12707 contains also changes in 12709).

[image]

haslinghuis, 2023-06-15:
Another issue on Discord reports this PR added 25 - 35% utilization (from 60 - 70 to 95%) on a X8 rendering the FC unusable as soon as lipo is attached while using configurator.

haslinghuis, 2023-06-15:
@limonspb so far I can tell issue only presented on BLHeli32.

Tom-PhotosDronesWebs, 2023-06-19:
I just put together a new build and have the same issue.
HGLRC Zeus F722 3-6S F7 Flight Controller, HGLRC Zeus 60A 3-6S 30x30 4in1 ESC
HGLRC Aeolus motors.
BetaFlight 4.4.2.
QUAD X
DSHOT 600, Bidirectional.

Zuldan, 2023-06-21:
Just did some testing...

4.4.0
- Errors on motors: **None**
- Motor direction working: **Yes**

4.4.2
- Errors on motors: **Yes, motor 4 within 1 minute**
- Motor direction working: **No**

4.4.3 (https://github.com/betaflight/betaflight/pull/12866)
- Errors on motors: **Yes, motor 4 within 1 minute**
- Motor direction working: **Yes**

[image]

haslinghuis, 2023-06-21:
@Zuldan please try BlHeli32 Rev 32.95

haslinghuis, 2023-06-21:
@SteveCEvans please investigate the telemetry errors.

limonspb, 2023-06-21:
> T-MOTOR F4 6A 1S AIO with Betaflight 4.4.2 and Bluejay 0.19.2 also reports telemetry errors

is that just 0.3% errors? What if you raise throttle to 100%? (props off, of course)

Zuldan, 2023-06-22:
> @Zuldan please try BlHeli32 Rev 32.95

Installed this today. Still have an error on motor 4. 

0.03 appears for 2 seconds then disappears. It comes back again after a minute or so then disappears.

I took the motors to 100% throttle a few times. Zero errors.

damosvil, 2023-06-22:
> > T-MOTOR F4 6A 1S AIO with Betaflight 4.4.2 and Bluejay 0.19.2 also reports telemetry errors
> 
> is that just 0.3% errors? What if you raise throttle to 100%? (props off, of course) I got AM32 ESC SkyStar AK55, that just shows 15% errors at full throttle in 4.4.2, 4.4.3 But Blheli 32 is ok.
> 
> Same AM32 ESC shows 0 errors in 4.4.0

Tested again in T-MOTOR F4 6A 1S AIO with Betaflight 4.4.2 and Bluejay 0.19.2. The 0.03% error is reported some time after enabling motor testing and then goes to 0%. At full throttle I saw a small glitch 0.03% on the errors but they went to 0 inmediately.

howels, 2023-07-29:
> If it's any help, on `master` from today, an an H730 FC, I see this with low RPM's:
> 
> [image]
> 
> At higher RPMs the error percentage goes to 0. ESCape32 ESC (rel5) firmware.

Can you point the Escape32 team to the dshot bible? https://brushlesswhoop.com/dshot-and-bidirectional-dshot/
Specifically the part about needing a 30us delay after the dshot commands for line switching before sending the dshot telemetry response.

haslinghuis, 2023-08-09:
@limonspb please test again with latest zulu firmware.

haslinghuis, 2023-08-31:
We don't get new data on this issue for a month. Closing now as it can be reopened if needed.

closed 2023-08-31 by haslinghuis