betaflight/betaflight #12863: meta quest 2 (laptop desktop) fpv/video over wifi

goofyseeker311, 2023-06-03:
### Is your feature request related to a problem? Please describe

no

### Describe the solution you'd like

default wifi fpv connection features included the chip options (headset/laptop fpv software required)

### Describe alternatives you've considered

there might be a separate video/control accessory pod that could attach to headset/laptop using the normal fpv video/control signal

sugaarK, 2023-06-03:
What???

Pancronos, 2023-06-04:
May be here https://github.com/fpv-wtf/voc-poc ? widefpv? cosmostreamer ?

ivan-ef, 2023-06-05:
> there might be a separate video/control accessory pod that could attach to headset/laptop using the normal fpv video/control signal

But there are receivers by eachine already, that send video to android both by cable or Wi-Fi. It's cheap. You just install their app on quest, test it and share your results here - profit. 

I guess the latency will be significant - 160-200ms vs 5ms on fpv googles. It will be ok for observation but may not be for flight. Eachine rotg 01, eachine g-model. Skydroid smartphone receiver. YouTube is your friend.

closed 2023-07-14 by github-actions[bot]