betaflight/betaflight #12868: OSD not working

Star4fish, 2023-06-04:
### Describe the bug

After update to 4.4.2 OSD freezes on first welcome screen 
BETAFLIGHT v.4.4.2
MENU...  etc

But cannot go to OSD menu using thr, yaw and pitch.
Even after arming OSD screen don't change, there is always "welcome" OSD screen on top even during flight, there is no proper OSD
on 4.4.1 everything is working ok apart from SD card but it's working on 4.4.2

### To Reproduce

Speedybee F405 v3 
update to 4.4.2 and turn on

### Expected behavior

"welcome" osd screen turn off and proper OSD with data turn on

### Support ID

```txt
17b578ff-b214-4f71-9d63-22ba5422b32d
```

### Flight controller

SPEEDYBEE F405 V3 stack

### Other components

speedybee tx800
geprc elrs nano
foxeer m10q 180 gps 5883
runcam pheonix 2 joshua edition

Star4fish, 2023-06-04:
[image]

haslinghuis, 2023-06-04:
Duplicate of #12583

closed 2023-06-04 by haslinghuis