betaflight/betaflight #12869: MSP communication two stop bit

camillo2008, 2023-06-05:
### Is your feature request related to a problem? Please describe

I have searched for a way to activate 2 stop bits communication over serial communication for MSP protocol, i would need this because i am using a Jetson orin to communicate, and i have noticed that in the [data sheet](https://developer.download.nvidia.com/assets/embedded/secure/jetson/orin_nx/docs/Jetson_Orin_NX_DS-10712-001_v0.5.pdf?eDHTjHuUE8byK1AXa-VDfzfDjxAIjSyGDJHHL_lSisIDTs5n62jN-KZGedQkz7TLeOOfnldvxdaaDlAa4xvEvOvJWZDdmpY6YZ1IbOAcnKJ9lRpcyai6WBFo_PQMHFatLUeYj8yLhBnnK8nW79MycUjshgGIFSjTCdOLwrCh6Zyyt2m7zRSfUXzJYz0bbg==&t=eyJscyI6ImdzZW8iLCJsc2QiOiJodHRwczovL3d3dy5nb29nbGUuY29tLyJ9) it says:
`The UART receiver input has low baud rate tolerance in 1-stop bit mode. External devices must use two stop
bits. In 1-stop bit mode, the UART receiver can lose sync between the receiver and the external transmitter
resulting in data errors/corruption. In 2-stop bit mode, the extra stop bit allows the UART receiver logic to align
properly with the UART transmitter.`

And since i am using baud rate 1000000 i think this is particularly important.

Does a change like this one break betaflight configurator?

### Describe the solution you'd like

Maybe including a flag to enable 2 stop bits instead of one.

### Describe alternatives you've considered

I have tried to use one stop bit, and it is working, sometimes it seems to lose the synchronization, i am not sure if this is due to this bit or is due to the fact that there is no hardware flow control, still the orin data sheet seems to suggest that.

ledvinap, 2024-12-03:
@amrathod: 
first test if 2 stopbit is really necessary. MSP has some resynchronization capability, so losing few packets (well under 1%) should not hurt. And writing robust drivers will eventually pay off.

It is trivial to to modify code - just add SERIAL_STOPBITS_2 on https://github.com/betaflight/betaflight/blob/master/src/main/msp/msp_serial.c#L68
But IMO it it not feature that should be part of upstream - it is not according to MSP protocol (And I'd prefer to avoid another SmartAudio madness).

camillo2008, 2024-12-17:
I did not investigate this issue further as no one else seemed to encounter problems; perhaps 1M is not as high as initially thought. However, since @ledvinap has provided a solution, I will proceed to close this issue.

closed 2024-12-17 by camillo2008