betaflight/betaflight #12872: iBus Telemetry not working (possibly cloud build issue?)

Schuwi, 2023-06-06:
### Describe the bug

The FC receives data from the remote control (FS-i6S) but the remote control does not receive any telemetry from the FC (only internal receiver telemetry: signal strength & voltage) even though telemetry is assigned to the port.
[image]

### To Reproduce

Flash cloud build (there is no IBUS telemetry option, only IBUS EXTENDED):
[image]

### Expected behavior

The FC telemetry should be sent to the remote control.

What I tested:
- I checked the iA10 SENS pin with an oscilloscope and could confirm (assumingly only) the iA10 was sending periodic discovery packets of `0x04 0x81 0xFA 0xFF` with no response.
- I found [this](https://github.com/betaflight/betaflight/blob/d8a9906e92ae3654d3eddfe30f6e0a169d088c64/src/main/cli/settings.c#L1263-L1265) code line and tested in CLI to see that `get ibus_sensor` would produce an error, leading me to believe that `USE_TELEMETRY_IBUS` was not defined.
- I used the cloud log to run an almost identical local docker build only adding the `-DUSE_TELEMETRY_IBUS` command line option and could confirm that now the telemetry works as expected.

### Support ID

```txt
94ee339b-a0b9-46cd-9643-5d23f854a787
```

### Flight controller

MAMBA F405 MK2

### Other components

Receiver: FlySky FS-iA10B
ESC: MAMBA F40 MK2

### How are the different components wired up (including port information)

iA10 + <-> 5V
iA10 - <-> GND
iA10 SENS <-> TX3
iA10 SERVO <-> SBUS (RX1)

ESC <-> FC (via shipped connector)

### Add any other context about the problem that you think might be relevant here

Cloud build log: https://build.betaflight.com/api/builds/cdf0d891e90dfa9908934cf4d7ce6d1c/log
Support ID of local build with added USE_TELEMETRY_IBUS define: `cd93f66c-10d0-4845-8279-89d70ec786dd`

haslinghuis, 2023-06-06:
Fixed in 4.5.0: https://github.com/betaflight/betaflight/pull/12541

haslinghuis, 2023-06-17:
Fixed in #12873 for 4.4.3

PR #12873 (2023-06-07): Fix USE_TELEMETRY_IBUS
Fixes: #12872 

Backported from: #12541
closed 2023-06-17 by haslinghuis