betaflight/betaflight #12874: DShoot 600 not working anymore

X-ForceFPV, 2023-06-08:
### Describe the bug

after coming from 4.4.1. and loading my settings, i noticed pid loop freq to be stuck at 4khz. 
looks like dshot can not be set above 300. if i select 600 and restart its back at 300.

### To Reproduce

just select dshot 600 and restart

### Expected behavior

accept dshot 600

### Support ID

```txt
966dfd9c-bbbe-4fe7-ae0b-7b13776b2bc5
```

### Flight controller

Flywoo GOKU Versatile F405 1-2S 12A AIO W / build-in ELRS 2.4g RX ICM42688

haslinghuis, 2023-06-08:
This is done here: https://github.com/betaflight/betaflight/blob/master/src/main/config/config.c#L676-L683

<p>Looptime and Performance Recommendation Table:</p>

MCU | IMU | Sampling Rate | Bidirectional DShot Status | PID Loop Rate | DShot Protocol
-- | -- | -- | -- | -- | --
F7X2, H7XX, G4XX, and similar | MPU60X0, ICM2060X, ICM42688P | 8 kHz | Enabled or Disabled | 8 kHz | DShot 600
  | BMI-270 | 3.2 kHz | Enabled or Disabled | 3.2 kHz | DShot 300
F405 | MPU6000, MPU6050, ICM20601, ICM42688P | 8 kHz | Enabled | 4 kHz | DShot 300
  | MPU6000, MPU6050, ICM20601, ICM42688P | 8 kHz | Disabled (not recommended) | 8 kHz | DShot 600
  | BMI-270 | 3.2 kHz | Enabled or Disabled | 3.2 kHz | DShot 300
F411 UART Rx ** | MPU6000, MPU6050, ICM20601 | 8 kHz | Enabled | 4 kHz | DShot 300
  | MPU6000, MPU6050, ICM20601 | 8 kHz | Disabled (not recommended) | 8 kHz | DShot 600
  | BMI-270 | 3.2 kHz | Enabled | 3.2 kHz | DShot 300
F411 SPI Rx *** | MPU6000, MPU6050, ICM20601 | 8 kHz | Enabled | 2 kHz | DShot 300
  | MPU6000, MPU6050, ICM20601 | 8 kHz | Disabled (not recommended) | 4 kHz | DShot 300
  | BMI-270 | 3.2 kHz | Enabled | 1.6 kHz | DShot 300
  | BMI-270 | 8 kHz | Disabled (not recommended) | 3.2 kHz | DShot 300

<p>**
 For F411 UART Rx applications, using both available UARTs AND enabling 
SoftSerial, Accelerometer, large numbers of OSD elements, and using a 
larger number of filters, stability may require lowering looprate to 
2kHz</p><p>*** There are no SPI Rx solutions that are strongly 
recommended for future development due to challenges in resource 
allocation and scheduler inconsistency that consistently emerge with SPI
 Rx designs.
Additionally, there are no RC ecosystems that are actively developing a 
supported SPI Rx solution (ExpressLRS 3.0 and later do not support SPI; 
FrSky does not support SPI Rx over any protocol, and other SPI Rx 
solutions have been fully deprecated).</p><p>Note that the use of gyros 
such as the BMI270 lowers the gyro loop rate from 8kHz to 3.2kHz and is 
therefore advantageous for F411 designs.</p>

X-ForceFPV, 2023-06-08:
Thank you for your reply.
Can you help me understand it more?

On 4.4.1. Dshot 600 works great with 8k/8k and Bidirectional DShot enabled!
I run only 3 rpm filters with a d-term biquad and oc'd to 192Mhz, CPU load is 55%.

Do i understand the confc.c correct and there is no way to keep my settings?
I don't see why enabling Bidirectional DShot should force DShot 300. Wouldn't a simple warning be more appropriate?

haslinghuis, 2023-06-08:
Yes it was done for stability reasons and avoiding issues due to much computation:

https://github.com/betaflight/betaflight/pull/11222

X-ForceFPV, 2023-06-08:
I would like to open a discussion about applying that to a F405.

It is quite a bit faster than a F411 and as you can see in my cpu load it is no problem.
Even without overclocking to 192Mhz the load is only 59%.

Please bring it back or at least let the user choose.

haslinghuis, 2023-06-08:
@SteveCEvans 
@TheIsotopes 

thoughts?

Quick-Flash, 2023-06-09:
> Yes it was done for stability reasons and avoiding issues due to much computation:
> 
> https://github.com/betaflight/betaflight/pull/11222

Faster dshot doesnt increase cpu load at all, faster pidloop does though. There really isn't a good reason not to allow dshot 600 on f411.

TheIsotopes, 2023-06-09:
DShot300 or DShot600 doesn't play a big role in cpu load that's right but once the rpm filter is set, the cpu load goes up rapidly. Overclocking is not a default and therefore not an topic in this discussion **(user settings at your own risk)**.

hence my opinion:
DShot600 and 8k/8k is no problem on f405 i fly it myself **but without rpmfilter**.
With DShot600 and rpmfilter overclocking is necessary for better cpu load and it is therefore up to the user to use it in this way.

imo the default f4xx should be left as they are now.

X-ForceFPV, 2023-06-09:
The F405 has enough power for all of that.
Overclocking on F405 is not needed to be around 60% CPU utilization with clean builds.

The problem is:
At the moment there is no way to get 8k/8k with DShot 600 and Bidirectional Dshot.

Its not about the default but the ability to use it at all.

TheIsotopes, 2023-06-09:
> The problem is:
> At the moment there is no way to get 8k/8k with DShot 600 and Bidirectional Dshot.

I understand when rpmfilter is turned on it's switches to 8k/4k and DShot300 and cannot be changed by the user.

@haslinghuis We should leave the setting as above as defaults but we should give the user the opportunity to change the settings anyway without guarantee and support.

X-ForceFPV, 2023-06-09:
> I understand when rpmfilter is turned on it's switches to 8k/4k and DShot300 and cannot be changed by the user.

You are probably right, i was not clear about the consequences of enabling rpmfilter as i was not aware about the mechanics behind the switches. 

> @X-ForceFPV would you be happy with that?

perfectly happy, thanks

TheIsotopes, 2023-06-09:
Ok, i'm just a respected tester and not a programmer so we should wait and see how @haslinghuis thinks about this and implements it.

X-ForceFPV, 2023-06-09:
I'm glad this was not introduced on F405 as announced in 4.3, but edited only recently in 4.4.2. by @SteveCEvans in #12709. Otherwise i would not have known that this is even possible.

Maybe you can add this to the changelog.

Adding to that i noticed that the G4 is also forcing DShot 300 contradicting the recommendations above.
Maybe that is interesing for G4 users as the G4 has slightly more MIPS than the F405.

"#if defined(STM32F4)|| defined(STM32G4)
        /* If bidirectional DSHOT is being used on an F4 or G4 then force DSHOT300. The motor update restrictions then applied"

jiri-jirus, 2023-06-11:
I tested current MAMBA F405 DJI MK2 with MPU6000, 8khz PID loop and RPM filter enabled. The osd behaved erraticaly. So i am definitely voting for ban of this setting.

haslinghuis, 2023-06-11:
[image]

X-ForceFPV, 2023-06-11:
> I tested current MAMBA F405 DJI MK2 with MPU6000, 8khz PID loop and RPM filter enabled. The osd behaved erraticaly. So i am definitely voting for ban of this setting.

thanks for testing.

i would like to know why you feel that way.

were you unable to select a working setting for yourself or do you just love the authoritarian way?
remember we are not talking about default settings. this should be best option for clean builds (maybe analog osd only?) and people who use blackbox

haslinghuis, 2023-06-11:
One way to handle the exception would be adding a custom define to override the reset here as this decision was made to reduce support issue when users are using parameters in out-of-spec boundaries.

X-ForceFPV, 2023-06-11:
some kind of obstacle to overcome like cli only (with or without a special mode) to keep most people from just selecting the biggest number would be effective and fine as long as there is a way.

nice benchmark btw. lets hope we get a ton of cheap and fast at32 controllers in the future.

haslinghuis, 2023-06-11:
For testing:

- Enable expert mode
- Enable show release candidates
- Select development
- Click Auto-detect or pick your target
- Select 4.5.0-zulu
- Add `EXPERTMODE` to custom defines
- Add `#12883` to commit field
- Load Firmware [Online]
- Flash Firmware

X-ForceFPV, 2023-06-11:
> For testing:
> 
> * Enable expert mode
> * Enable show release candidates
> * Select development
> * Click Auto-detect or pick your target
> * Select 4.5.0-zulu
> * Add `EXPERTMODE` to custom defines
> * Add `#12883` to commit field
> * Load Firmware [Online]
> * Flash Firmware

thanks!

when i had some time to test your changes, do i need to post the review in the #12883 or here?

SteveCEvans, 2023-06-11:
It’s not just about the CPU load. Debugging with SCHEDULER_DETERMINISM or TIMING_ACCURACY clearly shows that any attempt to run 8K PID loop on an F4 with bit-directional DSHOT, really messes up loop stability. You might think it “flies great” but the filtering is compromised and it flies better with a 4KHz PID loop. The BMI gyros are recommended for F4s because they reduce the load considerably as the gyro loop is also slower. It’s not hard to get an F4 totally unresponsive on USB if 8KHz PID loop is forced, necessitating a DFU re-FLASH. The restriction is there for good reason.

SupaflyFPV, 2023-06-17:
@SteveCEvans if we can show it running fine overclocked - can we have the option back via CLI command for those who want it?

if so I will give instructions to @X-ForceFPV

closed 2023-07-26 by github-actions[bot]

LordGG, 2023-11-03:
I know it is a bit late, but just wanted to add that I have no issue running F405 at 8K/8K DShot600 with RPM filter enabled. On the bench CPU Usage is 77%, and during a whole flight it stays between 83-84%.

I couldn't **feel** a difference, however, I have noticed when running 8K PID loop, setpoint is not clean, while it is absolutely smooth at 4K. Probably the side effect Steve was talking about (compromised RC Smoothing), thanks for the clarification :)