betaflight/betaflight #12875: Motor spin fast when the quad is armed (zero throttle)

dpatricksp, 2023-06-08:
### Describe the bug

On BF4.4 defaults, after arming motors go spinning like crazy (really fast).

### To Reproduce

On flight controller HAKRCF411D, after flashing BF4.4 motors spin erratically and fast just after arming at zero throttle.

### Expected behavior

expect the motors to run smooth 

### Support ID

```txt
# 

# dump

# version
# Betaflight / STM32F411 (S411) 4.4.1 Apr 11 2023 / 12:32:53 (e43d591b2) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HAKRCF411D
manufacturer_id HARC

# name: -

# resources
resource BEEPER 1 B02
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B06
resource MOTOR 4 B07
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A15
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B03
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 NONE
[... 970 lines cut ...]
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end
```

### Flight controller

HAKRCF411D

### Other components

VTX - Walksnail
RX/TX - TBS Tracer

cd334, 2023-06-09:
I have a similar problem.

I have a custom build 85mm whoop, with ex1103s 12kkv motors, happymodel X12 FC, betafpv 85mm frame and 2S batteries i have throttle limit to 70%, and also have similar issues, but not always.
Usually the with the first battery is OK, but on the second battery the after arm the motors spin fast, sometime so fast that the my drone instantly fly away, in other cases after giving some throttle the drone start to fly to the moon, only the disarm helps. After the problem started it is always there for a while.

I have tried many things but not helped:
- the problem is exist with bidirectional dshot ON and OFF
- not depend on the esc version and type (blheli_s official, latest bluejay)
- not depends on esc frequency (24k or 48k)
- not depend on battery compensation

Here are my cli dumps:
https://nc.czd.hu/s/sECR4y5E5WZMerC

haslinghuis, 2023-06-09:
> set dshot_bitbang = AUTO

check either ON or OFF

closed 2023-07-18 by github-actions[bot]