betaflight/betaflight #12891: Permanent RCSMOOTHING OSD warning

Schuwi, 2023-06-16:
### Describe the bug

The OSD is constantly showing a blinking "RCSMOOTHING" warning text while the craft is armed. Apparently this indicates that rc smoothing did not initialize properly. The rc smoothing option is disabled in the Receiver tab of the configurator.

### To Reproduce

1. Flash Betaflight v4.4.2
2. Disable RC Smoothing
3. Arm craft
4. Observe warning in OSD

### Expected behavior

There should be no warning about RC smoothing not working when it is not even enabled.

### Support ID

```txt
07fd08a1-1a56-4987-bfc7-95878870a540
```

### Flight controller

MAMBA F405 MK2

### Other components

Receiver: FlySky FS-iA10B
ESC: MAMBA F40 MK2
VTX: MATEKSYS VTX-HV
Camera: Foxeer HS1177

### How are the different components wired up (including port information)

iA10 + <-> 5V
iA10 - <-> GND
iA10 SENS <-> TX6
iA10 SERVO <-> SBUS (RX1)

VTX + <-> 9V
VTX - <-> GND
VTX TX <-> TX3
VTX Vid <-> Vid

Foxeer + <-> 5V
Foxeer - <-> GND
Foxeer VID <-> CAM

### Add any other context about the problem that you think might be relevant here

[This change](https://github.com/betaflight/betaflight/commit/636d563abe31378cb941fb6cebdc7eb36efd2d25#diff-72d835b20e40a1a7a44822a8142ee40c31c87a7628d5d544aadd1ab6be4ee258L889-R764) (link doesn't work properly: `src/main/fc/rc.c`, line 889/764) might have introduced the issue.

A workaround is to disable the RC Smoothing warning and Visual beeper in the OSD tab. Unless you have a beeper, which is probably also activated. I don't have one so I can't test that.

haslinghuis, 2023-06-17:
@Schuwi please test `#12893`

Schuwi, 2023-06-18:
@haslinghuis The (visual) beeper is not going off anymore but the "RCSMOOTHING" OSD warning is still there.
Probably missing the same change as in #12893 here: https://github.com/betaflight/betaflight/blob/a98364fa558460a6948db9c7248177071915448e/src/main/osd/osd_warnings.c#L380-L388

PR #12893 (2023-06-17): Fix rcSmoothing warning
Fixes #12891
PR #12898 (2023-06-19): Fix rcSmoothing warning OSD
Fixes #12891

Somehow missed fixing the OSD part in previous fix.
closed 2023-06-19 by haslinghuis