betaflight/betaflight #12894: H7 reboots ESC on every EEPROM save (4.4.x or 4.5 master)

limonspb, 2023-06-18:
### Describe the bug

Foxeer H7 20x20, Tmotor H7 mini, Foxeer H7 AIO (all H7 boards i have)
Every time user initiates EEPROM save it causes ESC to reboot (starting tones)
Ways to cause EEPROM save:
1. Change something on PID tab, hit save.
2. Change something in OSD menu, then select save and exit.
3. Set stats_min_armed_time_s = 1, arm the drone, disarm after a second.

little demonstration:
https://www.youtube.com/watch?v=-kPhtjUsgqg

### To Reproduce

1. Flash H7 to latest master
2. Connect ESC and motors (to hear the tones, props off)
3. Set protocol to DSHOT600
4. Connect battery
5. Open PID tab
6. Change some slider values
7. Hit save
8. You will hear ESC starting tones - not expected

### Expected behavior

No ESC starting tones should be heard on the regular saving.
ESC starting tones should present only at save and reboot.

### Support ID

```txt
13ed5ae0-d7c5-438b-9f8c-bc93dfaec321
```

### Flight controller

Tmotor H7 mini, Foxeer H7 20x20, Foxeer H7 AIO

### Other components

The bug reproducible with both Blheli_32, and AM_32.

SteveCEvans, 2023-06-18:
During a FLASH write the processor stalls and so there will be no DSHOT signals sent for a couple hundred ms or so. It is this loss of communication that's probably causing the ESC to reset. I'll try to reproduce to see if that's what's going on.

SteveCEvans, 2023-06-18:
950ms pause.

[image]

SteveCEvans, 2023-06-18:
On an F405 with the same ESC it's only 250ms and no restart beeps.

[image]

SteveCEvans, 2023-06-18:
I've proven this to be the cause using an F405 and the additional of a delay as shown below. 50ms delay doesn't results in beeps whereas 75ms does.
```
void writeEEPROM(void)
{
#ifdef USE_RX_SPI
    rxSpiStop(); // some rx spi protocols use hardware timer, which needs to be stopped before writing to eeprom
#endif
    systemConfigMutable()->configurationState = CONFIGURATION_STATE_CONFIGURED;

    delay(75);
    writeUnmodifiedConfigToEEPROM();
}
```

75ms shown below causing DSHOT to pause for 311ms.

[image]

haslinghuis, 2023-07-02:
Would be nice to get rid of this too:

[image]

SteveCEvans, 2023-08-23:
I believe this is for ESC firmware to resolve, not Betaflight.

limonspb, 2023-08-25:
> I believe this is for ESC firmware to resolve, not Betaflight.

Or if there is any desire to figure out why H7 eeprom save takes way longer than in F7/F4

hydra, 2023-09-26:
@SteveCEvans Did you confirm the H7 using an H743? Did you time the H730/H750 saving to an external flash, like the SPRacingH7RF/EXTREME/NANO/ZERO?

> Or if there is any desire to figure out why H7 eeprom save takes way longer than in F7/F4

a would say the H7 internal flash saving code needs checking and finding if the delay comes from the page erase or the writing of data.  H7's have a page erase size of 128k, which is quite large and probably where the delay is coming from.

hydra, 2023-09-26:
Also, the topic of this thread should probably be changed, as the ESCs don't actually reboot, AFAIK they just attempt to re-discover the ESC protocol again after a certain delay, which is generally what you want when configuring your FC on the bench otherwise you will have to power cycle the ESCs all the time...

Not to say that ESC authors can't do better though, such as making the motor beep when the signal is lost, making it double beep when the same signal is regained, or making it do the startup tones when a /different/ signal is detected.

haslinghuis, 2023-11-12:
Cannot reproduce ESC rebooting on H7EXTREME with master build ons Siskin_ST_32_Plus BLHeli32 32.8.

Also tried 32.10 with or without bidirectional dshot telemetry. Even able to change PID settings (like saving damping) while doing motor test. No RPM failures either.

But 32.10 gives another startup tune and now keeps beeping when bidirectional dshot is enabled on save and reboot.

hydra, 2023-11-13:
> Cannot reproduce ESC rebooting on H7EXTREME with master build ons Siskin_ST_32_Plus BLHeli32 32.8.

That was what I was expecting, as the H7EXTREME uses an external NAND flash for config storage which is quicker.  There'll still be a drop-out of the DSHOT signal, but it's short enough that the ESC firmware you tested doesn't do anything unusual.

Also, FYI, there's the `USE_CONFIG_IN_SDCARD` option, which may be slower than external NAND flash, also the H7EXTREME PX4 edition uses a NOR flash again with different timing characteristics.

IMHO this issue should be closed as there's nothing that the BF devs can do about it.  ESC firmware authors can make improvements, as per my previous comment: https://github.com/betaflight/betaflight/issues/12894#issuecomment-1735338938

nerdCopter, 2023-11-13:
@sskaug

closed 2023-11-13 by haslinghuis